openrazer/openrazer #1039: Support for [Tartarus Pro] 1532:0244

Voraga, 2020-03-03:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 004: ID 1532:0244 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0244 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     186
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0018  1x 24 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```
lrwxrwxrwx 1 root root  10 Mar  3 14:54 usb-Razer_Razer_Tartarus_Pro-event-if01 -> ../event28
lrwxrwxrwx 1 root root  10 Mar  3 14:54 usb-Razer_Razer_Tartarus_Pro-event-kbd -> ../event25
lrwxrwxrwx 1 root root  10 Mar  3 14:54 usb-Razer_Razer_Tartarus_Pro-if01-event-kbd -> ../event26
lrwxrwxrwx 1 root root  10 Mar  3 14:54 usb-Razer_Razer_Tartarus_Pro-if02-event-mouse -> ../event30
lrwxrwxrwx 1 root root   9 Mar  3 14:54 usb-Razer_Razer_Tartarus_Pro-if02-mouse -> ../mouse1

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
Setup is the same as the Tartarus Chroma, I don't have a screenshot but it is the exact same layout so I suspect that a copy pasta of the Chroma code would be a good place to start. 100% willing to run some tests. will start reading up on the above capture packets.

darix, 2020-03-07:
Doesnt it support binding different actions depending on how deep you press the key?

Voraga, 2020-03-08:
Yes, but in windows it's disabled by default. You need to go in and activate it for each key (I don't know if it's active but ignored prior to that.)

darix, 2020-03-08:
Well my point was that it would be nice to have the support for using that :)

darodve, 2020-05-07:
I also have the need for our Tartarus Pro drivers works in Openrazer. At this point, is there any way I can help ?, information, tests ... Thanks for the effort

TheYadda, 2020-05-26:
Also would appreciate this

darodve, 2020-05-26:
I'm web developer, but I haven't got experience developing drivers for linux systems, is there any information or documentation somewhere to help with the development?

chabad360, 2020-05-26:
For whoever captures a pcap, please include testing the variable depth keypress feature.

darodve, 2020-05-27:
Sorry for my ignorance, how can I capture the pcap?

chabad360, 2020-05-31:
@darodve have a look [here](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protoco)

Amdarin, 2020-06-07:
I just purchase one of it. This week when I get it at home I'm going to try to gather the information pcap and I will attach it. I hope it could help to get the device working in Linux

Syphdias, 2020-06-12:
> @darodve have a look [here](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protoco)

Fixed Link: @darodve have a look [here](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol)

chabad360, 2020-06-12:
whoops

Amdarin, 2020-06-13:
Hi all,

I need a help, because its the first time I gonna do it. I think I have ready all to capture packets. I followed the guide and This is the screenshot about my currently state.

[image]

The usbmon loaded. Virtual machine ready with tartarus pro plugged into virtual machine ( Win10). Wireshark open and the filter applied. I powered off the lights for tartarus. I don't find in wireshark the data fragment to apply like as column. I'm forgetting something to capture the packages? How can I get the data fragment?

Thank you

Amdarin, 2020-06-13:
Hi.

I attached the pcap of tartarus pro. The zip have pcap, txt with the process, and two sreenshots of synapse 3.0. I hope it can help to create the driver. Let me know if anyone need to gather other information.

[razer-tartarus-pro-pcap.zip](https://github.com/openrazer/openrazer/files/4775278/razer-tartarus-pro-pcap.zip)

chabad360, 2020-06-14:
@Amdarin Did you include captures for the half-presses?

chabad360, 2020-06-14:
> Well my point was that it would be nice to have the support for using that :)

@darix this may not be possible currently (at least according to a [stale kernel proposal](https://patchwork.kernel.org/patch/10549711/))...

Amdarin, 2020-06-14:
@chabad360 What you mean is capture pressing the key to the half? 
Pressing the key to the half I didn't but I can do if you need the captures or whatever other thing you need. Let me know if is it.

chabad360, 2020-06-14:
@Amdarin yes, that.

Amdarin, 2020-06-14:
@chabad360 Ok. I gonna take captured of it and as soon as possible I going to attach it here.

Amdarin, 2020-06-14:
@chabad360 Hi. Done. Let me know if you need any other thing.

[razer-tartarus-pro-half-pcap.zip](https://github.com/openrazer/openrazer/files/4777258/razer-tartarus-pro-half-pcap.zip)

Amdarin, 2020-06-16:
@chabad360  Hi again. Could you check it? It's fine or I have to make something more. Is it enought?

Regards

chabad360, 2020-06-16:
It's fine, but for now I'm not going to deal with it, as actually implementing this is much more complicated (unfortunately, only due to semantics)...

jeffclay, 2020-06-21:
@chabad360 are you working on support for this device without the half press feature? if so, do you need any help testing or anything? I'm running Fedora 32 and can dual boot to Windows 10 if you need anything.

chabad360, 2020-06-21:
@jeffclay I'm working on #1124, I needed to know about half-presses as it's relevant to that, so for now I don't need any help, but thanks for the offer!

TheZatherus, 2020-07-11:
Will be interested in the results of this as soon as it is ready.  I too have the Tartarus Pro and wanting it to work like it should and be able to reprogram the keys.

mjb187, 2020-09-04:
+1 interested in this as well. Since it's so similar to the Tartarus V2 (1532:022B) structurally, would it be possible to simply treat it as that device as a half-measure until full support is possible?

Geoffjnt, 2020-09-05:
> +1 interested in this as well. Since it's so similar to the Tartarus V2 (1532:022B) structurally, would it be possible to simply treat it as that device as a half-measure until full support is possible?

+1  interest here too.  Did you find a way to use the Tartarus V2 (1532:022B) driver?

TheZatherus, 2020-09-08:
Just to be clear, the Tartarus Pro works like a Keypad, but we cannot reprogram the keys.  The reprogramming certain or all keys is desirable

merger3, 2020-09-08:
I have a Tartarus Pro as well, and would love to control it without synapse. Maybe starting with something like RGB control so that programs like openRGB could start using it right away would be a good way to start?

BenjaminSonPham, 2020-10-08:
Has there been any update on this? Is there anything we could do to help?

lizrea, 2020-10-30:
Also interested in support for this. Not so worried about the half press feature, as I don't use it.

masc2279, 2020-11-29:
Would love support for the Tartarus PRO here is the dump from

lsusb -d 1532: -v

Bus 007 Device 004: ID 1532:0528 Razer USA, Ltd Razer BlackShark V2 Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0528 
  bcdDevice            1.00
  iManufacturer           3 
  iProduct                4 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0132
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              1 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0047
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 3
        bSourceID               6
        bControlSize            1
        bmaControls(0)       0x02
          Volume Control
[... 403 lines cut ...]
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

jbullara, 2021-01-21:
Has there been any update on support for this?

Amdarin, 2021-02-21:
Hello everyone,

Before asking I would like to thank all the people who are working on this project and helping to get the drivers for our devices. I hope that one day companies will give more support to their products for our beloved operating system. I would like to help you, but I am not a developer and I would like to ask you about updating this device. Do you think it's close to being released? I know the keybind profiles are far off and dependent on other equipment, but that would be great when you're done.

Also I would like to ask if there is any other way to help this project for the people who are not developer

Thank you and good job.

z3ntu, 2021-02-22:
FYI even without openrazer support you should be able to use e.g. https://github.com/sezanzeb/key-mapper to remap the keys, which is probably what most of you are looking for. Also please give feedback whether that application solves your use cases.

Amdarin, 2021-03-05:
Hi guys,

@z3ntu I hav
I have tried and I had some difficults to map the keys. Someones of them work properly but other not too much well. It's not full what we need but it's for a few time useful.

I hope the fully support of these kind of devices someday be a reality because each day in bussiness envirements linux is dominating but in desktop enviroments always is pending.

Thank you for your support

Moosatronic, 2021-03-11:
for anyone looking for tartarus pro support it looks like currently keybinding support is being added at https://github.com/openrazer/openrazer/pull/1124 to openrazer and then afterwards we will probably see stuff like the tartarus pro v2 being added afterwards where you can get support both for lighting and rebinding.

Moosatronic, 2021-03-12:
that said, also if the developers need more packet capture I also have a tartarus pro I can capture packets from

sezanzeb, 2021-03-13:
> I have tried and I had some difficults to map the keys. Someones of them work properly but other not too much well. It's not full what we need but it's for a few time useful.

If there are any problems with key-mapper I'd love to fix them. Please post an issue here https://github.com/sezanzeb/key-mapper/issues and describe what went wrong.

Amdarin, 2021-03-15:
@sezanzeb 

I was trying much more and I know what was happening. For ejemple the device have by default a pre-assigned keys, i just give you an example. 1 Key and 2 key have "1" and "2" preassigned so if you assign the same key by error or the first time you use the software and you don't know that it could happend then if you pressing the key 1 or 2  you will get a "1" or "2" repeting without stop. "1111111111111111111"....  It's not an exactly a error but perhaps could show any message warning at the same moment of the setting up the keyboard. I was testing it with World of Warcraft and I can tell you the feeling and respond of the keyboard was pretty good. Also it could be good if we have profiles but I think it could be far of the focus project. I would like comment too that I have a razer basilisk V2 and one key of it is not reconize but I going to post it on the web project. Thank you for you support and effort..

jerrygb, 2021-05-02:
Hi @chabad360

Irrespective of the half presses, would we have support for lighting and full key presses for this device?

Thanks

mercedes7935, 2021-05-22:
What’s the current stand? I would like to buy a Tartarus pro but learned from mistake with my leadr and search before, but the Tartarus isn’t supported yet for linux, is it?

TransLunarInjection, 2021-11-25:
@mercedes7935 I'm currently using a Tartarus Pro on linux without openrazer.

It's usable as a basic keyboard, I don't have a way to switch it to joystick inputs.

You can use [sezanzeb/key-mapper](/sezanzeb/key-mapper) to remap inputs.

[image]

Looking forward to getting support in openrazer eventually.

Amdarin, 2022-02-06:
Hi all,

Is there any news about this device to get properly handle of lights?

With key-mapper I can bind properly the keys although there are someone keys which not work, I don't know if the boys of the key-mapper project can fix it.

Anyway, its the only device I left to get all my devices working properly Basilisk V2 mouse and Huntsman elite keyboard, and this is the last one.

I wish know how to programming to help you but I can't.

Regards

ksignorini, 2022-02-27:
Any update on this ever happening?

z3ntu, 2022-03-26:
There's an open PR (#1577) but it needs additional work to be in a mergable state.

justinkb, 2022-06-05:
I'm probably buying one in a week or so, can help assist in testing then

justinkb, 2022-09-09:
I have this device now, how can I help get this working properly?

hcarter1112, 2022-09-27:
I currently have one of these and would love to help get this working. Please let me know what you need. I can't code but if I have some instructions, I am willing to help. =) Thanks to every dedicated member of this community! It is amazing that we have opensource support at all for these devices! =)

Amdarin, 2022-12-31:
Hi all.

I was trying OpenRGB and it seem it works with all peripherics. I think thank you to your hard works. But I discovered gratefully that Tartarus pro rgb works here and it's detected. I posted it because can help you to implemented it and polichromatic software can detect it and work with it. I could change the RGB lights effect easily and with many effects. Perhaps you can shared information with the other developer team to get it implemented here.

Regards
[image]

stlhrt, 2023-01-12:
[tartarus_pro_options_and_effects.pcapng.gz](https://github.com/openrazer/openrazer/files/10397280/tartarus_pro_options_and_effects.pcapng.gz)
Fresh pcap of effects and settings changes.

SolacianAllie, 2024-07-16:
I hope this gets support someday, Id love to finally have some control over the thing on linux

z3ntu, 2024-08-31:
Somebody will need to finish #1817, then it'll be supported :)

extrakerned, 2025-04-26:
Throwing another +1 for this device, which has become very popular over the last few years.

Jennigma, 2025-12-11:
Tartarus Pro support is the last piece I need to move from Windows to Linux. I use the Tartarus for assistive device reasons, so I'm hoping this gets sorted soon. 

Thanks for working on this marcusfairclough-maker !

MicHaeL-MonStaR, 2025-12-12:
> Tartarus Pro support is the last piece I need to move from Windows to Linux. I use the Tartarus for assistive device reasons, so I'm hoping this gets sorted soon.
> 
> Thanks for working on this marcusfairclough-maker !

Can you explain what you need it to do but doesn't on Linux?

I can tell you that, if you need it to be remap-able, input-remapper does work for that. - It just doesn't do some of the "fancier" things like on-the-fly profile-switching or make use of the analog switched. - Or, at least, there seem to be some settings for that, but if that should work with the Tartarus Pro, I didn't figure out how. - Perhaps that's something it needs drivers for to be recognized as analog switches/inputs.

But in any case, other than those things, all keys and buttons and scroll-wheel work, as I couldn't "live without" this device on PC anymore either, having used the SpeedPad by Belkin since about 2005, then moved to this one. - So, I mean, if I couldn't get it to work, that would've been a big drawback moving to Linux for me as well, but luckily it wasn't much of an issue.
I'd still like to see more bespoke support, I suppose, perhaps also some more features in input-remapper, but just letting you know that works if you didn't know about it.