openrazer/openrazer #1041: Support for 1532;024B, 2019 advanced model with optical keyboard

779g, 2020-03-08:
Hello to all,

Thank you for your hard work and for making the drivers for razer available in linux.

I would like to ask whether the model 1532:024b (2019 advanced 15 inch model with optical keyboard) is supported, or whether there are plans ( it would be possible ) to support it in the future. It doesnt appear on the list of the homepage of the github project.

For reference I am currently  using the drivers version 2.7.0 and loaded them using `sudo modprobe razerkbd`

I tried to follow https://github.com/openrazer/openrazer/wiki/Setting-up-the-keyboard-driver, but I'm stuck at the binding stage with the following message

```
root:/sys/bus/hid/drivers/razerkbd# echo -n "0003:1532:024B.0004" > bind      
-bash: echo: write error: No such device
```

For information, I have 
```
adrian@aztlan:/sys/bus/hid/drivers/hid-generic$ ls -l
total 0
lrwxrwxrwx 1 root root    0 mar  8 12:47 0003:0461:0010.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:0461:0010.0005
lrwxrwxrwx 1 root root    0 mar  8 12:47 0003:0461:0010.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:0461:0010.0006
lrwxrwxrwx 1 root root    0 mar  8 12:47 0003:1532:024B.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:024B.0002
lrwxrwxrwx 1 root root    0 mar  8 12:47 0003:1532:024B.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:024B.0003
lrwxrwxrwx 1 root root    0 mar  8 13:02 0003:1532:024B.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:024B.0004
--w------- 1 root root 4096 mar  8 13:02 bind
lrwxrwxrwx 1 root root    0 mar  8 12:47 module -> ../../../../module/hid_generic
--w------- 1 root root 4096 mar  8 12:47 new_id
--w------- 1 root root 4096 mar  8 12:23 uevent
--w------- 1 root root 4096 mar  8 12:57 unbind
```

with the device ending in 004 being the mouse device.

Thank you for the help and for your time

779g, 2020-03-21:
I will provide a pcap in the next days. I think I understood the procedure, as described in https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

779g, 2020-04-15:
@z3ntu I have made this capture following the guide. I have only done the static mode not to overcrowd the capture with too much information. You can rename the following file to pcapng

I have toggled on and off the static mode to none - #ff1234 - none - #4411ff

If you need a capture with more mode feel free to reach and I'll be more than happy to help.

Thanks !

[staticmode-ff1234-4411ff-bladeadv-1532-024b.txt](https://github.com/openrazer/openrazer/files/4482312/staticmode-ff1234-4411ff-bladeadv-1532-024b.txt)

z3ntu, 2020-04-19:
pcaps are good, thanks.

Your device uses `0x1f` as transaction_id and the `razer_chroma_standard_matrix_effect_*` functions. Unfortunately with new devices it's a game of trial and error, the pcaps only reveal the bare essentials these days.

779g, 2020-04-19:
Thank you. Is there something more I could do to help ?

z3ntu, 2020-04-19:
It's very likely that Synapse 2 doesn't support your device, but yes with v2 we got proper infos from the captures, Synapse 3 emulates nearly everything in software which is annoying.

You can of course try implementing support yourself, see e.g. https://github.com/openrazer/openrazer/commit/e934da922ca8e327b669e014f88a1d222270e170 as an example.

779g, 2020-04-19:
@z3ntu 

If I understand correctly (I'm going into the details because i'm completely new to driver pgoramming and C)

1) See if synapse 2 supports the device, if so do another capture with it
2) If synapse 2 supports the device, it might use the transaction id 0x80 which would make us able to simply add the device as a new case to the devices like on e934da9  
3) if it's the case, I can simply follow the referenced commit and to the PR
3.1) So I fork, clone the repo, I branch it out
3.2) I make the same changes as in the referenced PR, make install it, modprobe it (?) and then test the driver using a head like polychromatic or razergenie ?
4) if this works then submit the commit for PR

Is this correct ?

If we cannot capture the keyboard config with a 0x80 code, what are the implications ? How to rev-eng it, and what do you mean by trial and error ?

z3ntu, 2020-05-01:
It is a fair bit of trial and error unfortunately. Not much we can do on our side.

napsterred, 2021-10-09:
Hey i couldn't figure out whether  to reverse engineer the usb or edit the device id in open razer . Could anyone please post instructions that work for the RB15 ADVANCED late 2019 optical switch model. I tried all these steps didn't know which to focus on to successfully make the keyboard recognizable. Thanks for  support for all the models guys, If ya could help me with this one it would be a charm , :24B , thanks guys.

napsterred, 2021-12-03:
Is there any data i can provide to make the process successful please?

z3ntu, 2021-12-04:
@napsterred As written a few comments up, you can try to do the implementation yourself based on the info in https://github.com/openrazer/openrazer/issues/1041#issuecomment-616114657 and trial & error. Or you wait until another person with this device does that. But spamming the comments doesn't help anybody.

napsterred, 2021-12-04:
i have that device , do you guys need new pcap ??

napsterred, 2021-12-13:
> 

Hi could ya please tell me how manprobe the device generated kernal or where the kernel after make file will be installed and how?

z3ntu, 2021-12-13:
@napsterred Please stop spamming the issue tracker. Nothing will happen by you writing these comments.

napsterred, 2022-01-03:
incorporate , updated device support for 24B

https://github.com/napsterred/openrazer.github.io

napsterred, 2023-04-16:
Hey guys, was the device id updated in the new openrazer kernal, if not i can provide an updated kbrddriver.c file ? If not, what additional data is required to get it to work. Thanks, much appreciate your efforts.

PR #1737 (2024-04-06): Add support for Razer Blade 15 Advanced (Late 2019)
Changes extracted from #1727. Credit goes to @napsterred.

I don't own this device, so can't confirm 100% working, but the changes look good to me.

Closes #1727.
Fixes #1041 (under the assumption the optical keyboard already works or is out of scope for OpenRazer?)
closed 2024-04-06 by z3ntu