openrazer/openrazer #1045: Same as "Unable to control Tartarus #500"

coberry87, 2020-03-15:
### Distribution

openrazer-daemon/bionic,bionic,now 2.7.0~ubuntu18.04.1 all [installed,automatic]
openrazer-driver-dkms/bionic,bionic,now 2.7.0~ubuntu18.04.1 all [installed,automatic]
openrazer-meta/bionic,bionic,now 2.7.0~ubuntu18.04.1 all [installed]
python3-openrazer/bionic,bionic,now 2.7.0~ubuntu18.04.1 all [installed,automatic]
razergenie/unknown,now 0.8.1-1 amd64 [installed]
polychromatic/bionic,bionic,now 0.3.12-bionic1 all [installed]

### Support

- [x ] Have you tried rebooting?
- [x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I am experiencing the same problem as in issue #500 (Once I set it to pulse, I am no longer able to set it back to static) and since there was no resolution from that issue, I am opening up this one for a continuation of that one. All my results line up just like the other user's minus the bus location of course. What do you need from me to move forward?

coberry87, 2020-04-19:
Can I please get an update? It has been over a month.

z3ntu, 2020-04-24:
@coberry87 No idea what the problem is there. It works on all other devices so it's not something I can debug on my own devices. The usb packet captures I have look completely normal so it doesn't seem like there's something special going on in Synapse that we don't do.

closed 2020-10-04 by stale[bot]