openrazer/openrazer #1047: Support for Basilisk X Hyperspeed

ShadowBullet, 2020-03-16:
Adding support for the Basilisk X Hyperspeed would be very appreciated

z3ntu, 2020-03-16:
Please fill out the template you can find at https://github.com/openrazer/openrazer/issues/new?assignees=&labels=Device+Support&template=new-device-support.md&title=Support+for+%5Bdevice+name+here%5D (you can edit this issue or open a new one, however you wish) with the information listed there, otherwise we can't do much at all.

bno1, 2020-04-19:
### Device Info

* Name: Razer Basilisk X Hyperspeed
* Product Number: RC30-03150100-0000
* Model Number: RC30-031501

### Device Information
<details>
    <summary>lsusb -d 1532: -v</summary>

    Bus 001 Device 002: ID 1532:0083 Razer USA, Ltd Razer Basilisk X HyperSpeed
    Device Descriptor:
      bLength                18
      bDescriptorType         1
      bcdUSB               2.00
      bDeviceClass            0 
      bDeviceSubClass         0 
      bDeviceProtocol         0 
      bMaxPacketSize0        64
      idVendor           0x1532 Razer USA, Ltd
      idProduct          0x0083 
      bcdDevice            2.00
      iManufacturer           1 Razer
      iProduct                2 Razer Basilisk X HyperSpeed
      iSerial                 0 
      bNumConfigurations      1
      Configuration Descriptor:
        bLength                 9
        bDescriptorType         2
        wTotalLength       0x0054
        bNumInterfaces          3
        bConfigurationValue     1
        iConfiguration          0 
        bmAttributes         0xa0
          (Bus Powered)
          Remote Wakeup
        MaxPower              500mA
        Interface Descriptor:
          bLength                 9
          bDescriptorType         4
          bInterfaceNumber        0
          bAlternateSetting       0
          bNumEndpoints           1
          bInterfaceClass         3 Human Interface Device
          bInterfaceSubClass      1 Boot Interface Subclass
          bInterfaceProtocol      2 Mouse
          iInterface              0 
            HID Device Descriptor:
              bLength                 9
              bDescriptorType        33
              bcdHID               1.11
              bCountryCode            0 Not supported
              bNumDescriptors         1
              bDescriptorType        34 Report
              wDescriptorLength      94
             Report Descriptors: 
               ** UNAVAILABLE **
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x81  EP 1 IN
            bmAttributes            3
              Transfer Type            Interrupt
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0040  1x 64 bytes
            bInterval               1
        Interface Descriptor:
          bLength                 9
          bDescriptorType         4
          bInterfaceNumber        1
          bAlternateSetting       0
          bNumEndpoints           1
          bInterfaceClass         3 Human Interface Device
          bInterfaceSubClass      0 
          bInterfaceProtocol      1 Keyboard
          iInterface              0 
            HID Device Descriptor:
              bLength                 9
              bDescriptorType        33
              bcdHID               1.11
              bCountryCode            0 Not supported
              bNumDescriptors         1
              bDescriptorType        34 Report
              wDescriptorLength     159
             Report Descriptors: 
               ** UNAVAILABLE **
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x82  EP 2 IN
            bmAttributes            3
              Transfer Type            Interrupt
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0040  1x 64 bytes
            bInterval               1
        Interface Descriptor:
          bLength                 9
          bDescriptorType         4
          bInterfaceNumber        2
          bAlternateSetting       0
          bNumEndpoints           1
          bInterfaceClass         3 Human Interface Device
          bInterfaceSubClass      0 
          bInterfaceProtocol      1 Keyboard
          iInterface              0 
            HID Device Descriptor:
              bLength                 9
              bDescriptorType        33
              bcdHID               1.11
              bCountryCode            0 Not supported
              bNumDescriptors         1
              bDescriptorType        34 Report
              wDescriptorLength      61
             Report Descriptors: 
               ** UNAVAILABLE **
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x83  EP 3 IN
            bmAttributes            3
              Transfer Type            Interrupt
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0040  1x 64 bytes
            bInterval               1
    Device Status:     0x0000
      (Bus Powered)
</details>

<details>
    <summary>ls -lah /dev/input/by-id/</summary>

```
total 0
drwxr-xr-x 2 root root 160 Apr 19 15:04 .
drwxr-xr-x 4 root root 600 Apr 19 15:04 ..
lrwxrwxrwx 1 root root  10 Apr 19 14:46 usb-Chicony_Electronics_Co._Ltd._Lenovo_EasyCamera_0001-event-if00 -> ../event14
lrwxrwxrwx 1 root root  10 Apr 19 15:04 usb-Razer_Razer_Basilisk_X_HyperSpeed-event-if01 -> ../event11
lrwxrwxrwx 1 root root   9 Apr 19 15:04 usb-Razer_Razer_Basilisk_X_HyperSpeed-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Apr 19 15:04 usb-Razer_Razer_Basilisk_X_HyperSpeed-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root  10 Apr 19 15:04 usb-Razer_Razer_Basilisk_X_HyperSpeed-if02-event-kbd -> ../event12
lrwxrwxrwx 1 root root   9 Apr 19 15:04 usb-Razer_Razer_Basilisk_X_HyperSpeed-mouse -> ../mouse0
```
</details>

### Packet Captures
[basilisk_x_hyperspeed.zip](https://github.com/openrazer/openrazer/files/4498921/basilisk_x_hyperspeed.zip)

### Synapse Screenshots
<details>
<summary>Screenshots</summary>

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

</details>

bno1, 2020-04-29:
Is anyone working on this? If not, I can give it a shot.

bno1, 2020-05-01:
This mouse doesn't have support for `charge_status`, it says that the request is unknown:
```
[ 4980.515096] razermouse: Command not supported. Start Marker: 05 id: ff Num Params: 02 Reserved: 07 Command: 84 Params: 00000000000000000000000000000000 
```

However, some user-space programs (https://github.com/polychromatic/polychromatic/issues/246) try to use it and not implementing it would cause errors in them. From the response to that issue it looks like they can't check if the device supports `charge_status`/`is_charging`, but only if it has a battery or not.

A simple solution would be to fake `charge_status` for this device to always return 0. Here is a patch that works for me:

```diff
diff --git a/driver/razermouse_driver.c b/driver/razermouse_driver.c
index ae07c2d..50c436f 100644
--- a/driver/razermouse_driver.c
+++ b/driver/razermouse_driver.c
@@ -736,10 +736,18 @@ static ssize_t razer_attr_read_get_battery(struct device *dev, struct device_att
  */
 static ssize_t razer_attr_read_is_charging(struct device *dev, struct device_attribute *attr, char *buf)
 {
-    struct usb_interface *intf = to_usb_interface(dev->parent);
-    struct usb_device *usb_dev = interface_to_usbdev(intf);
+    struct razer_mouse_device *device = dev_get_drvdata(dev);
     struct razer_report report = razer_chroma_misc_get_charging_status();
-    struct razer_report response_report = razer_send_payload(usb_dev, &report);
+    struct razer_report response_report = {0};
+
+    switch(device->usb_pid) {
+    // Wireless mice that don't support is_charging
+    case USB_DEVICE_ID_RAZER_BASILISK_X_HYPERSPEED:
+        return sprintf(buf, "0\n");
+        break;
+    }
+
+    response_report = razer_send_payload(device->usb_dev, &report);

     return sprintf(buf, "%d\n", response_report.arguments[1]);
 }
```

Does this sound reasonable? Is it ok to add this to my pull request?

BNolet, 2020-05-11:
Not a dev but a user of this mouse. I'd be okay with this as the device is technically always not charging and discharging.

josealmeida13, 2020-05-28:
Any update on this?

bno1, 2020-05-28:
I have a PR pending. I used it almost every day since I started working on it, so I'd say it's fine if you are willing to build it from source.

brandon-feder, 2020-05-29:
Hello everyone. I just got the Basilisk X Hyperspeed and want to build this repository from source. However, as I am fairly knew to Linux in general, I was hoping someone could give instructions or point to a good resource on how to compile this and use it with other software such as RazerGenie. I don't mean to waste anyone's time but if anyone can spare a minute or two, I could use some help. Thanks!

BNolet, 2020-05-30:
Replace the git clone step in the section on [this](https://github.com/openrazer/openrazer/wiki/Building-a-package) page for your Linux distro with the following:

`git clone https://github.com/bno1/openrazer.git -b feature_basilisk_x_hyperspeed`

Otherwise follow the guide

brandon-feder, 2020-05-30:
Thanks you for your response. I am running Arch Linux which in the instructions say to just get the package from the user repository. However, the package is not yet merged and is not in the AUR. Are there any other resources on how to compile it in Arch? Thanks you for your time.

bno1, 2020-05-30:
For Arch you can edit this package: https://aur.archlinux.org/packages/openrazer-meta-git/.
Clone it, then open the PKGBUILD file in an editor and replace the `source` line with this:
```
source=("git+https://github.com/bno1/openrazer.git#branch=feature_basilisk_x_hyperspeed")
```

The run `makepkg -sC`. It should install all the dependencies and build the packages in the same directory. Install them with `pacman -U`.

BNolet, 2020-05-30:
@bno1 Thanks for posting this. I'm personally not running Arch but if you're willing to, I think it would be much appreciated if you edited [this](https://github.com/openrazer/openrazer/wiki/Building-a-package) page to include more detailed build steps like the other distros listed.

bno1, 2020-05-30:
@BNolet I'd be more than happy to contribute, but I don't know how to edit that page or propose changes to it. How does this work?

BNolet, 2020-05-30:
🤔 Hmm, it seems like there's a few ways but all of them require action on the part of the maintainers for this repo. @z3ntu, do you think you might be able to give @bno1 access to edit the Wiki for this purpose?

bno1, 2020-06-01:
@BNolet it seems to be a limitation of the mouse. The driver sends the X and Y dpi, but the mouse seems to ignore the Y dpi and use X dpi for both axis. Also, Razer Synapse doesn't have controls for separate X and Y dpi.

brandon-feder, 2020-06-01:
Thanks bno1 for your help. That was exactly what I needed.

josealmeida13, 2020-06-03:
I've built it from source on Arch Linux and everything has been working pretty well. Thanks to @bno1.

z3ntu, 2020-06-12:
@bno1 If you have suggestions for the wiki page, just send them as a comment and I'll integrate it into the wiki page. I've had to set the wiki read-only because of spam issues and I don't think I can give wiki access to specific users without giving them any other permissions.

bno1, 2020-06-13:
@z3ntu 

----

# Arch Linux

## Install from AUR
Install the open-razer-meta or the open-razer-meta-git package from AUR.

## Build packages from a different github repository
This is useful if you want to build the openrazer packages from a fork of the main repository and/or from a specific git branch or commit.

1. Git clone the [open-razer-meta-git](https://aur.archlinux.org/packages/openrazer-meta-git/) AUR package by going to its page and using the "Git Clone URL":
    ```
    git clone https://aur.archlinux.org/openrazer-git.git
    ```
2. Inside the openrazer-git directory there should e a PKGBUILD file. Inside it there is a line that defines the git source to use: `source=("git+https://github.com/openrazer/openrazer.git")`. You can edit this in one of the following ways:
    * Edit this to use another git repo. For example, use your own fork of the openrazer repo:
        ```
        source=("git+https://github.com/<your github username>/openrazer.git")
        ```
    * Edit this to use a specific branch:
        ```
        source=("git+https://<git repo url>#branch=<branch name>")
        ```
    * Edit this to use a specific commit:
        ```
        source=("git+https://<git repo url>#commit=<commit hash>")
        ```
3. Run `makepkg -sC` in the PKGBUILD directory to build the openrazer packages. This will also install any needed dependencies.
4. If the build was successful, the packages should be in the PKGBUILD directory with the extension `.pkg.tar.gz` or `.pkg.tar.zst`. Install them with `pacman -U`, for example:
    ```
    pacman -U ./*openrazer*.pkg.tar.*
    ```

## Build packages from a source directory
This is useful if you are developing a feature and you want to build from the local source tree that you're working on.

The steps are the same as in the previous section except for step 2. Instead of editing the `source` line, do the following and replace the `/path/to/your/openrazer/source/tree` with your own:
```diff
--- a/PKGBUILD
+++ b/PKGBUILD
@@ -11,8 +11,12 @@ arch=('any')
 url="https://github.com/openrazer/openrazer"
 license=('GPL2')
 makedepends=('git' 'python-setuptools')
-source=("git+https://github.com/openrazer/openrazer.git")
-sha256sums=('SKIP')
+source=()
+sha256sums=()
+
+prepare() {
+    cp -ar "/path/to/your/openrazer/source/tree" "$srcdir/openrazer"
+}

 pkgver() {
   cd "$_pkgbase"

```

This will copy your local openrazer source tree to the src directory used by makepkg instead of cloning it from git.

qnm, 2020-06-25:
I've installed this on Pop!_OS 20.04 (Ubuntu-ish) and it's working great - except that Polychromatic alerting me that my mouse is at zero charge. TY @bno1

bno1, 2020-06-25:
If the notification happens when the mouse is sleeping or turned off it's going to show 0%.

Martomate, 2020-08-15:
Any progress on this? There is a PR, but is it ready for merge?

bno1, 2020-08-23:
Yes, in bluetooth mode the `hid-generic` driver is used instead of `razermouse`. I don't know what changes are necessary to make bluetooth mice work with openrazer unfortunately.

closed 2020-09-26 by z3ntu

c2xusnpq6, 2020-12-13:
`RC30-03150100-0000` = `RZ01-03150100-R3A1` ?

c2xusnpq6, 2020-12-13:
[image]