openrazer/openrazer #1050: Support for BlackWidow Tournament 2014

sech1p, 2020-03-19:
### Device Info

* Name: BlackWidow Tournament 2014
* Product Number: RZ03-0081
* Model Number: RZ03-00810900-R3M1

### Device Information
```
Bus 003 Device 002: ID 1532:011c Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x011c 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
```

```
total 0
drwxr-xr-x 2 root root 280 mar 19 12:13 .
drwxr-xr-x 4 root root 660 mar 19 12:13 ..
lrwxrwxrwx 1 root root  10 mar 19 10:58 usb-18f8_USB_OPTICAL_MOUSE-event-if01 -> ../event10
lrwxrwxrwx 1 root root   9 mar 19 10:58 usb-18f8_USB_OPTICAL_MOUSE-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 mar 19 10:58 usb-18f8_USB_OPTICAL_MOUSE-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root   9 mar 19 10:58 usb-18f8_USB_OPTICAL_MOUSE-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 mar 19 12:13 usb-Razer_Razer_BlackWidow_Tournament_Edition-event-if01 -> ../event5
lrwxrwxrwx 1 root root   9 mar 19 12:13 usb-Razer_Razer_BlackWidow_Tournament_Edition-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 mar 19 12:13 usb-Razer_Razer_BlackWidow_Tournament_Edition-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 mar 19 12:13 usb-Razer_Razer_BlackWidow_Tournament_Edition-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 mar 19 12:13 usb-Razer_Razer_BlackWidow_Tournament_Edition-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 mar 19 10:58 usb-Trust_Webcam_Trust_Webcam_Trust_Webcam-event-if00 -> ../event22
lrwxrwxrwx 1 root root  10 mar 19 10:58 usb-Wacom_Co._Ltd._MTE-450-event-mouse -> ../event11
lrwxrwxrwx 1 root root   6 mar 19 10:58 usb-Wacom_Co._Ltd._MTE-450-mouse -> ../js0
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
Currently unreachable because I have no way to do a VM to run WireShark.
### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
Currently unreachable because I have no way to do a VM to run Synapse..

z3ntu, 2020-03-19:
Actually a duplicate of #55 but let's use this one for tracking this device.

blshkv, 2020-06-13:
[Razer_BlackWidow_Tournament_2014.tar.gz](https://github.com/openrazer/openrazer/files/4773810/Razer_BlackWidow_Tournament_2014.tar.gz)

Hello everyone. I have captured all events (I think):
 - change brightness of logo;
 - change gamemode led;
 - record macros
 - keys

Do let me know if I missed anything

z3ntu, 2020-07-07:
@blshkv Thanks! 
If you want to get your hands dirty with the code, here's roughly what's needed: 

In `driver/razerkbd_driver.h` add `#define USB_DEVICE_ID_RAZER_BLACKWIDOW_TE_2014 0x011C`
In `driver/razerkbd_driver.c` add your device (`USB_DEVICE_ID_RAZER_BLACKWIDOW_TE_2014`) everywhere where you can find `USB_DEVICE_ID_RAZER_BLACKWIDOW_ULTIMATE_2012` (for device name add a new case statement of course)
Add pid `011c` to `install_files/udev/99-razer.rules`
In `daemon/openrazer_daemon/hardware/keyboards.py` copy-paste blackwidow ultimate 2012 block and rename to blackwidow te 2014
Add to readme
Run `./scripts/generate_appstream_file.sh`

And test if it works of course :smile: 

If you need any help, either comment here (probably slow) or message me via email/telegram/matrix

blshkv, 2020-07-07:
I was kind of hoping you will do it blindly and we will test it 😅

z3ntu, 2020-07-11:
@blshkv Done :wink: See #1194

blshkv, 2020-07-16:
Hi, just wanted to update that I tested the basic function (which is logo led pretty much) and it works. However, KeyWatcher does not start and GameMode and Micro recoding does not work. Here is the debug log:

```
2020-07-16 08:38:04 | razer                          | INFO     | Found device.0: 0003:1532:011C.000B
2020-07-16 08:38:04 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowTournamentEdition2014
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.setDeviceMode method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDriverVersion method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.getSerial method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceMode method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasDedicatedMacroKeys method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.getRazerUrls method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.getKeyboardLayout method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.setBrightness method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.led.gamemode.getGameMode method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.led.gamemode.setGameMode method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.setMacroMode method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.getMacroMode method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.getMacroEffect method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.setMacroEffect method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.lighting.bw2013.getEffect method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.lighting.bw2013.setPulsate method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.lighting.bw2013.setStatic method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.macro.getMacros method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.macro.deleteMacro method to DBus
2020-07-16 08:38:04 | razer.device0                  | DEBUG    | Adding razer.device.macro.addMacro method to DBus
2020-07-16 08:38:04 | razer.device0.keymanager       | WARNING  | No event files for KeyWatcher
2020-07-16 08:38:04 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality

```

blshkv, 2020-07-16:
ok, found the typo:
https://github.com/openrazer/openrazer/pull/1194/files#diff-54cfdf20ba8b5ba5526fd387938a6a37R412

PR #1194 (2021-06-30): Add support for Razer BlackWidow Tournament Edition 2014
Fixes #1050

Building from source instructions can be found here: https://github.com/openrazer/openrazer/wiki/Building-a-package

@blshkv @sech1p
closed 2021-06-30 by z3ntu