openrazer/openrazer #1053: Linux Mint 19 RazerBlackWidowElite Issues

BonitaTerror, 2020-03-22:
## Distribution
Linux Mint 19 Tara
### Support

- [Yes] Have you tried rebooting?
- [Yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I am using polychromatic controller as a front end to modify my keyboard settings. Occasionally, it will work without a hitch, but quite frequently I will run into issues that polychromatic reports as likely a bug by the openrazer daemon. I thought I would just give some info here and see what others think.

When launching polychromatic I am greeted with the following error:

[image]

I'm sure there are other logs to look into for info, but from all I can find there seem to be no other error reports

I am appreciative of any help you may be able to offer!

z3ntu, 2020-03-22:
Can you please run `hexdump -C /sys/bus/hid/drivers/razerkbd/0*/device_serial` ? It looks like you have another device where Razer failed to program the serial number correctly. Weirdly the existing code to handle these cases didn't seem to trigger for you.

BonitaTerror, 2020-03-22:
Sure!

Here is the output:
```
alexander@totoro:~$ hexdump -C /sys/bus/hid/drivers/razerkbd/0003\:1532\:0228.0003/device_serial 
00000000  49 4f 31 39 34 30 46 32  36 35 30 30 30 31 39 0a  |IO1940F26500019.|
00000010
```

z3ntu, 2020-03-23:
So you mean it does work now?

BonitaTerror, 2020-03-24:
It seems to randomly work and break. Would it be absurd to assume this could be a hardware issue?

z3ntu, 2020-03-24:
Do you see many messages containing `razerkbd` in the dmesg?

BonitaTerror, 2020-03-24:
A few, but only 1 really seems like it could be problematic

```
alexander@totoro:~$ dmesg | grep razerkbd
[    2.292262] razerkbd: loading out-of-tree module taints kernel.
[    2.292283] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
[    2.720213] razerkbd 0003:1532:0228.0001: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer BlackWidow Elite] on usb-0000:00:14.0-2/input0
[    2.780191] razerkbd 0003:1532:0228.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow Elite] on usb-0000:00:14.0-2/input1
[    2.840216] razerkbd 0003:1532:0228.0003: input,hidraw2: USB HID v1.11 Mouse [Razer Razer BlackWidow Elite] on usb-0000:00:14.0-2/input2
```

closed 2020-08-24 by stale[bot]