openrazer/openrazer #1056: Deamon cannot be relaunched (Pyolochromatic Issue) Manjaro Linux

Mathew1602, 2020-03-23:
I am not to sure why it is not working and i cant get it to launch 

here is my error from the  terminal :

[image]

z3ntu, 2020-03-23:
Hi, could you please give me the output of the command `lsusb | grep 1532` and of the command `sudo modprobe razerkbd`? Thanks.

lah7, 2020-03-23:
There is a typo in the command - should be a lowercase `v`, like so:

    openrazer-daemon -Fv

Mathew1602, 2020-03-23:
> Hi, could you please give me the output of the command `lsusb | grep 1532` and of the command `sudo modprobe razerkbd`? Thanks.

Here are the commands you asked me to run, and thank you for helping me so far :)

[image]

z3ntu, 2020-03-23:
It looks like you forgot to install the kernel headers for your kernel. You can install them with `sudo pacman -S linux55-headers` and then reboot.

Mathew1602, 2020-03-23:
> It looks like you forgot to install the kernel headers for your kernel. You can install them with `sudo pacman -S linux55-headers` and then reboot.

Hello so i installed the headers and then rebooted but now i get this error instead. 

[image]

Mathew1602, 2020-03-23:
> It looks like you forgot to install the kernel headers for your kernel. You can install them with `sudo pacman -S linux55-headers` and then reboot.

When i try to add the user then log out and log back it I still get the same error 

[image]

lah7, 2020-03-23:
Try this instead - and re-log in:

    sudo gpasswd -a $USER plugdev

Mathew1602, 2020-03-23:
> Try this instead - and re-log in:
> 
> 
> 
>     sudo gpasswd -a $USER plugdev

Thank you that worked :), I really appreciate the help. I tried to get this to work for days. Thank you again

closed 2020-03-23 by z3ntu