openrazer/openrazer #1057: MOK issue

leepeffer, 2020-03-23:
Hi, 

Hopefully this isn't a super dumb question. How can I check if the installation was successful after rebooting? I don't see any new apps and I cant seem to find the daemon everyone is talking about.

I am trying to install openrazer on Elementary Os Hera and at the end of the installation a bright pink window popped up asking for me to define a password and that it would ask for that password again after reboot. I must have missed the part where it asked again because I restarted without inputing that password anytime. 

What is that all about?

Sorry, new to Linux. Thanks!

lah7, 2020-03-23:
Hello, welcome! You can check OpenRazer is running by looking for a `openrazer-daemon` process running in the background. elementaryOS doesn't ship a **System Monitor** by default, so you'd have to install one in **AppCenter** or open the Terminal:

    ps -aux | grep openrazer-daemon

:point_up: _This should list two processes - one is the actual process, and the other is because of the `grep` command._

If you need to diagnose the daemon, run the daemon like so:

    openrazer-daemon -Fv

---

OpenRazer on its own is just a **backend**, so if you have a keyboard, the gaming mode and on-the-fly macro functions will now start to function.

To complement OpenRazer, you'll need to choose a **frontend**. You can find these on [OpenRazer's homepage](https://openrazer.github.io) under "Applications", like [Polychromatic](https://polychromatic.app/) or [RazerGenie](https://github.com/z3ntu/RazerGenie).

Note that Polychromatic has a tray applet, but elementaryOS 5.0 removed support for them, so [there's some extra steps needed to get it back](https://github.com/polychromatic/polychromatic/issues/187#issuecomment-441374353).

**elementaryOS** is based on **Ubuntu**.

---

This bright pink thing:

[image]

... isn't actually related to your password. This adds your user account to the group `plugdev`, so an application like OpenRazer can take control of the peripherals. Typing `groups` in the terminal will show you what else your user is part of - for instance `sudo` lets you run administrative commands.

leepeffer, 2020-03-23:
Thanks for the response!

When I run your grep command, I only get the following one line returned.

`23168  0.0  0.0  40044  1088 pts/1    S+   09:56   0:00 grep --color=auto openrazer-daemon
`

The '-Fv' command returns a large list of drivers / devices.

Also, the big pink popup that I was referring to what literally a huge popup that took over the whole terminal. It mentioned something about "MOK."

I got RazerGenie working. It can't find any devices. I'm on a Razer 15 (2018) laptop and I am assuming at least the keyboard should show up.

How would be the proper way to delete all openrazer files and try another fresh install?

z3ntu, 2020-03-23:
From what I can find on the internet, this message seems to be about Secure Boot and is probably triggered by DKMS. The easiest solution is to turn off Secure Boot in your BIOS/UEFI.

leepeffer, 2020-03-23:
Ok, I'll do that. Regarding reinstalling openrazer. Can I simply just do the installation process again safely, or should I remove whatever fragments may have been left behind with the failed install?

z3ntu, 2020-03-23:
Once Secure Boot is off and you still have the packages installed, you can run 
`sudo dkms install openrazer-driver/2.7.0` to install the kernel driver again and you should be good to go.

leepeffer, 2020-03-23:
Done! Thanks guys. Its up and running and now I can see my laptop on RazerGenie.

z3ntu, 2020-03-23:
Great, glad you got it working!

closed 2020-03-23 by z3ntu