openrazer/openrazer #1059: Device Not Recognized

HanumanH, 2020-03-23:
Hi,
My device is not recognized even though everything in troubleshooting guide seem to be all cleared. 

Kernel Modules output is:
Module openrazer-driver/2.7.0 already installed on kernel 5.5.10-arch1-1/x86_64 

Loading the kernel module did not give me any errors. 

Starting the daemon with openrazer-daemon –Fv give me this output: 
Unable to lock on the pidfile. 
I have already added user to group plugdev and rebooted.
 
I have restarted the system many times. I also try various solutions from the forums and google but does not seem to help. 

My device is listed on the Device Support page: 
Razer Tartarus V2  1532:022B 

The output of lsusb | grep '1532:' is 
Bus 003 Device 002: ID 1532:022b Razer USA, Ltd Razer 

I have used Polychromatic Controller, razerCommander and RazerGenie each of them give me a message of: Device Not Recognised, No device connected and No device was detected respectively. 

Please kindly help me with some possible solution. 
Greatly appreciate your help.

HanumanH, 2020-03-23:
My System Info:
Arch Linux 5.5.10-arch1-1
Awesome Window Manager
LigthDM Display Manager with Webkit Theme Aether as greeter

z3ntu, 2020-03-23:
Hi, thanks for the detailed description what you've tried! The problem in your case is that your device is not supported in the 2.7.0 version yet but will be included in the next release. So in the meantime you can install `openrazer-git` from the AUR.

HanumanH, 2020-03-23:
I don't really know where to get openrazer-git
Searching AUR on arch linux give me 10 different packages.
Are you saying that I should use openrazer-daemon-git version 2.6.0.r6.g2433ae8-1?

Search Result:
openrazer-daemon | 2.7.0-1 |
openrazer-daemon-git | 2.6.0.r6.g2433ae8-1 |
openrazer-driver-dkms | 2.7.0-1 |
openrazer-driver-dkms-git | 2.6.0.r6.g2433ae8-1 |
openrazer-meta | 2.7.0-1 |
openrazer-meta-git | 2.6.0.r6.g2433ae8-1 |
python-openrazer | 2.7.0-1 |
python-openrazer-git | 2.6.0.r6.g2433ae8-1 |
razergenie | 0.8.1-1 |
razergenie-git | 0.7.r8.ga786269-1 |

z3ntu, 2020-03-23:
OpenRazer is split into 4 packages in the AUR, the kernel driver, the daemon, the python bindings and a meta package. The so-called pkgbase is openrazer-git: https://aur.archlinux.org/pkgbase/openrazer-git/

HanumanH, 2020-03-23:
Thank you. I'm going to install this package and report back soon.

HanumanH, 2020-03-23:
I spend two working days; 8 hours a day trying to fix this problem on my own with no luck.
With only 2 comments from @z3ntu; the misery is over for me. Thank you so much @z3ntu!

A note for other Linux noobs like myself that might have the same problems as I; after receiving the direction from @z3ntu I did the following:
1. Check which driver version I have:
# sudo dkms status openrazer-driver
2. Uninstall the driver that I have
# sudo dkms uninstall openrazer-driver/$version
$version is the version that was listed in the result of step one.
# reboot
4. Download and install
# git clone https://aur.archlinux.org/openrazer-git.git
# cd openrazer-git
# makepkg -si
# reboot
5. Open Polychromatic Controller and my device is in the window. :-)

z3ntu, 2020-03-24:
No problem. My general advice is: If you don't get it working yourself, spend some time (but not 2 days :wink: ) trying to work it out yourself and if you don't manage it, just ask for help and show that you actually tried to get it working, went through the troubleshooting guide etc (as you did here). There are a handful of very common setup issues with openrazer and yours was one of them :)
Closing the issue then.

closed 2020-03-24 by z3ntu