openrazer/openrazer #1060: Support for Razer Turret for Xbox One

tkeinan, 2020-03-24:
### Device Info

* Name: Razer Turret for Xbox One
* Product Number: R203-02820100-0000
* Model Number: R201-0282

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 010: ID 1532:0904 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0904 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     179
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 160 Mar 24 17:54 .
drwxr-xr-x 4 root root 620 Mar 24 17:54 ..
lrwxrwxrwx 1 root root  10 Mar 24 17:28 usb-CNFEH63M3520100009E2_Integrated_Webcam_HD-event-if00 -> ../event15
lrwxrwxrwx 1 root root  10 Mar 24 17:54 usb-Razer_Razer_Turret_Receiver-event-if01 -> ../event10
lrwxrwxrwx 1 root root  10 Mar 24 17:54 usb-Razer_Razer_Turret_Receiver-event-if02 -> ../event11
lrwxrwxrwx 1 root root   9 Mar 24 17:54 usb-Razer_Razer_Turret_Receiver-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Mar 24 17:54 usb-Razer_Razer_Turret_Receiver-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Mar 24 17:54 usb-Razer_Razer_Turret_Receiver-mouse -> ../mouse1

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

tkeinan, 2020-03-24:
Will update by the end of the week to include:
* Packet capture
* Screenshots of Synapse

free-vbucks, 2020-09-28:
Pictures in Synapse **for the keyboard**
Toolbar-
[image]
Customize-
[image]
Customize, continued, with sidebar out-
[image]
Lighting-
[image]
Power-
[image]

free-vbucks, 2020-11-25:
Here is the packet capture, a key thing about this device is that _both the mouse and keyboard come in through the same USB dongle_. If requested, I can capture them individually using the charging cable.

[turret.pcapng.zip](https://github.com/openrazer/openrazer/files/5599190/turret.pcapng.zip)

free-vbucks, 2020-11-25:
I used a rainbow "wheel" effect, like the wave but pivoting on a center instead of moving in a line

z3ntu, 2020-11-29:
From the pcaps:

* The mouse uses transaction id 0x1f and extended custom frame (`razer_chroma_extended_matrix_effect_custom_frame`) with matrix dimensions [1, 1]
* The keyboard uses transaction id 0x9f and extended custom frame (`razer_chroma_extended_matrix_effect_custom_frame`) with matrix dimensions [6, 18]

But I'm not sure we can support this device well at the moment because it's essentially two devices going over the same usb device and we definitely don't have a device in openrazer yet that does that

quidamphx, 2024-04-01:
Is there any hope of support for this being added?

DantehZe, 2026-01-04:
> [@quidamphx](https://github.com/quidamphx) Somebody who has the device needs to add support for it, see https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md for details.

I tried to do this myself but I honestly have no idea what I'm doing, even with the instructions.