openrazer/openrazer #1064: Open Razer Crashes as soon as it is launched via razer Genie.

paulwin1, 2020-03-27:
openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2020-03-27 19:44:54 KST; 24s ago
       Docs: man:openrazer-daemon(8)
    Process: 34500 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
   Main PID: 34500 (code=exited, status=1/FAILURE)

 3월 27 19:44:54 ManjaroDesktop systemd[1296]: Starting Daemon to manage razer devices in userspace...
 3월 27 19:44:54 ManjaroDesktop openrazer-daemon[34500]: Starting daemon.
 3월 27 19:44:54 ManjaroDesktop openrazer-daemon[34500]: 2020-03-27 19:44:54 | razer                          | CRITICAL | User is not a member of the plugdev group
 3월 27 19:44:54 ManjaroDesktop openrazer-daemon[34500]: Stopping daemon.
 3월 27 19:44:54 ManjaroDesktop systemd[1296]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
 3월 27 19:44:54 ManjaroDesktop systemd[1296]: openrazer-daemon.service: Failed with result 'exit-code'.
 3월 27 19:44:54 ManjaroDesktop systemd[1296]: Failed to start Daemon to manage razer devices in userspace.

Using Manjaro Linux Gnome. Currently on 5.4 LTS kernel using stable packages.

lah7, 2020-03-27:
> CRITICAL | User is not a member of the plugdev group

Please run this command, and re-log in:

    sudo gpasswd -a $USER plugdev

closed 2020-09-07 by stale[bot]

paulwin1, 2020-12-31:
I apologize for the hiatus. There have been events in my life I would not like to get into.

I have fixed the issue by following the steps on the trouble shooting page.