openrazer/openrazer #1069: Falha no RazerGenie

brunno2020, 2020-03-29:
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Sun 2020-03-29 10:28:48 -03; 22s ago
     Docs: man:openrazer-daemon(8)
  Process: 17692 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 17692 (code=exited, status=1/FAILURE)

brunno2020, 2020-03-29:
Alguma solução?

z3ntu, 2020-03-29:
Hi, it's likely that your user is not a member of the plugdev group. You can add it by running `sudo gpasswd -a $USER plugdev` and reboot. Then it should work fine :)

Also English would be preferred as I don't speak your language.

closed 2020-09-07 by stale[bot]