openrazer/openrazer #1071: How to install on Deepin

woann, 2020-03-30:
How to install on Deepin

z3ntu, 2020-03-30:
Deepin seems to use dpkg and apt, so you can run `./scripts/build_debs.sh` from the repository to build some deb files which you can install. In case you get an error, check the file linked in the script output, it'll likely be some missing dependencies. Also it might not build at all on Deepin in case the dependencies are named differently.

woann, 2020-03-30:
@z3ntu ok thinks

closed 2020-03-30 by woann

z3ntu, 2020-03-30:
According to #645 you might be able to use the Debian Sid repository (linked on our website openrazer.github.io)

woann, 2020-03-30:
I already tried,But Debian Sid repository  is not found.Maby It could be remove or discard.

woann, 2020-03-30:
@z3ntu I already tried,But Debian Sid repository is not found.Maby It could be remove or discard.