openrazer/openrazer #1073: "Device Not Recognised" "User not in plugdev", even though user is

Dxhs, 2020-03-31:
Kernel: 5.4.28-1-MANJARO

Installed Openrazer according to the Arch Wiki.
https://wiki.archlinux.org/index.php/Razer_peripherals#OpenRazer

Output of `sudo lsusb`:
`Bus 001 Device 005: ID 1532:006c Razer USA, Ltd`

Output of `sudo dkms status`:
`openrazer-driver, 2.7.0, 5.4.28-1-MANJARO, x86_64: installed`

Members of group `plugdev`:
`plugdev:x:982:eiriklade,sddm` (first user is me)

Output of `openrazer-daemon -Fv`:
`Unable to lock on the pidfile.`

The first 8 lines of `.../razer.log` is the same repeated:
`| CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!`

And the last 11 (which I'm guessing is useful?):
`| DEBUG    | Adding razer.devices.getSyncEffects method to DBus`
`| DEBUG    | Adding razer.devices.supportedDevices method to DBus`
`| DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus`
`| DEBUG    | Adding razer.daemon.stop method to DBus`
`| DEBUG    | Adding razer.daemon.version method to DBus`
`| DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus`
`| DEBUG    | Adding razer.devices.syncEffects method to DBus`
`| DEBUG    | Adding razer.devices.getDevices method to DBus`
`| INFO     | Serving DBus`
`| DEBUG    | DBus called get_serial_list`
`| DEBUG    | DBus called get_serial_list`

The mouse I'm struggling with is the Mamba Elite.

z3ntu, 2020-03-31:
The Mamba Elite is not yet supported in the latest stable OpenRazer release (I'm guessing you have that installed). Until the next release you can install `openrazer-git` from the AUR or wait for the next release which should be released in the coming weeks.

Dxhs, 2020-04-01:
@z3ntu What's the difference between the `-git` packages and the others? It seems the `-git` package you're referring to is outdated in the AUR being (as of writing) version 2.6.0.r6.g2433ae8-1 last updated 2019-11-10 12:30 and the non-git package at 2.7.0-1 last updated 2020-01-11 10:36.

Btw, there is no `openrazer-git` as far as I'm aware. Check this out:

https://aur.archlinux.org/packages/?O=0&SeB=nd&K=openrazer&outdated=&SB=n&SO=a&PP=50&do_Search=Go

z3ntu, 2020-04-01:
The `-git` package builds from latest git master (the pkgver doesn't update automatically on the AUR web interface but generally all `-git` packages always build from the latest source). And `openrazer-git` is the pkgbase and not the name of an individual package: https://aur.archlinux.org/pkgbase/openrazer-git/ , that's why you don't find it by searching for "Name, Description"

Dxhs, 2020-04-02:
Aah I see. I tried a complete install of the `-git` packages and the mouse was detected. My confusion came from reading up on the git page then installing the non-git packages. I see now the webpage for the the stable versions don't list the Mamba Elite yes. My mistake. Appreciate the help

Dxhs, 2020-04-02:
Aah I see. I tried a complete install of the -git packages and the mouse was detected. My confusion came from reading up on the git page then installing the non-git packages. I see now the webpage for the the stable versions don't list the Mamba Elite yes. My mistake. Appreciate the help

closed 2020-04-02 by Dxhs