openrazer/openrazer #1076: Support for [Razer Kraken Ultimate]

AndLouis, 2020-03-31:
### Device Info

* Name: Razer Kraken Ultimate
* Product Number: 1532:0527
* Model Number: RZ04-03180100-R3M1

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 002: ID 1532:0527 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0527 
  bcdDevice           50.02
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          291
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength          114
        bInCollection           2
        baInterfaceNr( 0)       1
        baInterfaceNr( 1)       2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID            10
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               6 
[... 258 lines cut ...]
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0025  1x 37 bytes
        bInterval
```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 160 apr  1 00:28 .
drwxr-xr-x 4 root root 600 apr  1 00:28 ..
lrwxrwxrwx 1 root root  10 apr  1 00:28 usb-Azurewave_USB_Camera_0x0001-event-if00 -> ../event14
lrwxrwxrwx 1 root root  10 apr  1 00:28 usb-Razer_Razer_Blade-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 apr  1 00:28 usb-Razer_Razer_Blade-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 apr  1 00:28 usb-Razer_Razer_Blade-if02-event-mouse -> ../event12
lrwxrwxrwx 1 root root   9 apr  1 00:28 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 apr  1 00:28 usb-Razer_Razer_Kraken_Ultimate_00000000-event-if03 -> ../event6

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

tariq86, 2020-04-30:
I'm interested in getting this device supported too...getting everything set up for a pcap now, I'll add it here when I'm done.

z3ntu, 2020-05-02:
@tariq86 If you have packet captures from the device, I'd appreciate if you could share them.

tariq86, 2020-05-02:
Here are the few that I've captured: [kraken-ultimate__pcaps.zip](https://github.com/openrazer/openrazer/files/4568619/kraken-ultimate__pcaps.zip)

Naming convention of pcaps:
 * The first part is the feature being captured (i.e. `lighting_`, `volume_`, etc.)
 * The second part is the function being captured (i.e. `breathing`, `spectrum-cycling`, etc.)
 * Some of them include a multiplier at the end, indicating the # of times I performed the action (i.e. `_x3`).

So, for example, file `lighting_brightness-0-to-100_x3.pcapng` is the PCAP of me switching the lighting brightness from 0 to 100 a total of 3 times.

Please let me know if these aren't good, or if you need anything else! I've only used Wireshark for capturing internal network traffic before, I've never used it for USB monitoring.

z3ntu, 2020-05-03:
I don't see any packets in there but if everything in #1131 works properly, it's fine :)

PR #1131 (2020-05-07): Add support for Kraken Ultimate [headset]
#### Fixes #1076 

This PR adds basic support for the [Kraken Ultimate](https://www.razer.com/gaming-audio/razer-kraken-ultimate) headset.

I tested static, breathing 1-3, spectrum, and on/off via [RazerGenie](https://github.com/z3ntu/RazerGenie), and they are all working as intended.
closed 2020-05-07 by z3ntu