openrazer/openrazer #1077: Orbweaver Chroma key remapping on daily

schaeggi, 2020-04-01:
My recent scripts using xkbcomp did stop working correctly for one key when I switched to daily. Then I noticed that the keycode name of the Main mod key, the one above the stick, changed from <LALT> to <LCTL>. Is this on purpose? This might lead to problems with the workflow of some people.
[Installed on Xubuntu18.04 from ppa:openrazer/daily as well as built from git.]

z3ntu, 2020-04-01:
I can't think of any change that we've done in the upcoming release other than https://github.com/openrazer/openrazer/commit/bd864c87c64908611cc989109c4970c14cf314fb that could've broken something in that way. Can you try the stable packages again and see if it's still the same behavior? I'm guessing an unrelated package was upgraded on the way and that it's not caused by the openrazer upgrade :smile:

schaeggi, 2020-04-02:
If it didn't change since yesterday, the stable package from ppa:openrazer/stable still uses <LALT>. I switched a few times between the stable and daily release to confirm.

z3ntu, 2020-04-02:
Can you try reverting the commit I've linked above (https://github.com/openrazer/openrazer/commit/bd864c87c64908611cc989109c4970c14cf314fb) on master, rebuild the kernel modules (using `make`) and then inserting the new driver (`rmmod razerkbd; insmod ./driver/razerkbd.ko` I think)?

schaeggi, 2020-04-02:
Reverted the commit and installed the complete package again as inserting the driver lead to an operation not permitted error.
With the reverted commit xev gave me the old Alt_L back, otherwise Control_L. I double checked.

z3ntu, 2020-04-03:
Interesting.. Can you add some debugging to the driver and rebuild again, as described in https://github.com/openrazer/openrazer/pull/873#issuecomment-577804267 ?

Then send me only the relevant parts from the dmesg from when you pressed this button?

schaeggi, 2020-04-03:
[   77.071750] razerkbd event: size 8 - data
[   77.071758]  razerkbd: 04 00 00 00 00 00 00 00
[   77.231750] razerkbd event: size 8 - data
[   77.231759]  razerkbd: 00 00 00 00 00 00 00 00
[   77.466747] razerkbd event: size 8 - data
[   77.466755]  razerkbd: 04 00 00 00 00 00 00 00
[   77.651747] razerkbd event: size 8 - data
[   77.651755]  razerkbd: 00 00 00 00 00 00 00 00

z3ntu, 2020-04-05:
So as far as I can tell, the code now modifies the `04 00 00 00 00 00 00 00` events to become `01 00 00 00 00 00 00 00` and not much more really. 

> the first byte [0] is 0x04 (which means "game controller")

But let's come from the other side: is it not logical to have this button be Control_L and not Alt_L? As far as I can tell from the pictures of that device, the button is just a random button which the device has. I have no clue what the button is commonly used for, but I'm guessing that Ctrl vs. Alt doesn't make much of a difference.

If we keep this, yes it'll be a breakage but I think people can live with it - the same as they currently live with it being Alt_L?

schaeggi, 2020-04-05:
Well I do think that Control_L is not that useful there because it's pressed with the thumb. So the spacebar can not be pressed at the same time. But the default keymap of this device is pretty much useless anyway (no escape key etc.). 
All in I I don't think that its that big a deal, as long as you know it. It took me a while to realize,why my scripts didn't do what they did before.
No idea if I'm wrong here, but I think the intended purpose of this button could be to switch between different keyboard profiles.

z3ntu, 2020-04-06:
Okay I'll add a note to the release notes. Thanks for bringing up this issue!

closed 2020-04-06 by z3ntu

chabad360, 2020-04-08:
One plus I've noticed from this change is that `Control_L` can be held, while `Alt_L` can't, I use this button as my backspace, so I no longer need to repeatedly press the button for it to work that way.

z3ntu, 2020-05-02:
@schaeggi  The change has been reverted on master as the same issue was also present on normal keyboards (see #1132)