openrazer/openrazer #1078: Support for Razer Basilisk V2

MatheusJCastro, 2020-04-01:
I didn't find an open issue for Basilisk V2, just for the previous version. So here we go:

### Device Info

* Name: Razer Basilisk V2
* Product Number: RZ01-03160100
* Model Number: RZ01-0316
### Device Information
Outpu of `lsusb -d 1532: -v`:
```
    Bus 001 Device 014: ID 1532:0085 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0085 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

Output of `ls -lah /dev/input/by-id/`:
```
lrwxrwxrwx 1 root root   9 abr  1 15:44 usb-Razer_Razer_Basilisk_V2-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 abr  1 15:44 usb-Razer_Razer_Basilisk_V2-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 abr  1 15:44 usb-Razer_Razer_Basilisk_V2-if02-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 abr  1 15:44 usb-Razer_Razer_Basilisk_V2-mouse -> ../mouse0

```

### Packet Captures

I didn't understand what exactly captures do you need. What I did here is just switch from light off to static. Hope it works.

[static.pcapng.txt](https://github.com/openrazer/openrazer/files/4417335/static.pcapng.txt)

### Synapse Screenshots

[image]
[image]
[image]
[image]

MatheusJCastro, 2020-04-01:
@z3ntu I uploaded the `.pcapng`. Why did you add the label `need pcap`? Is the file wrong?

z3ntu, 2020-04-01:
Sorry I missed that. pcaps are fine :)
What I can tell you from the file:
* Transaction id `0x1f`
* `razer_chroma_extended_matrix_brightness(VARSTORE, ZERO_LED, brightness);` is being used
* `razer_chroma_extended_matrix_effect_custom_frame()` is being used with row 0, start 0 and end 1 (which should results in `MATRIX_DIMS = [1, 1]`)

oinkbark, 2020-04-20:
Should be easy now if #804fa58 checks out

mihak1989, 2020-04-26:
Thank you MatheusJCastro for creating the request for Basilisk V2 and others as well for relevant post. I went through code upgrades for Basilisk in 804fa58: doing the similar exercise would not be so difficult, however Basilisk V2 has some new functionalities (https://www.youtube.com/watch?v=yHoQQpEyHQU) and I do not know how to address them properly. I have become aware  of openrazer project two days ago, so I believe it would take me days or weeks to do the proper job.

If anyone knows the solution I would be extremely grateful.

malsyned, 2020-05-04:
I went through and made a patch for the V2 that is a pretty straight clone of 804fa58. Basic testing shows that it works for e.g. setting DPI and controlling the wheel and logo LEDs.

The next feature I want to tackle is getting the side tilt on the scroll wheel to work. They default to "Repeat Scroll Up" and "Repeat Scroll Down", which "works" but isn't what I want those buttons to do. If I use Synapse to set the left and right tilt to "Scroll Left" and "Scroll Right" respectively, It doesn't work. More specifics:

Before I added V2 support to openrazer, usbmon didn't even show any button reports at all when I pressed those buttons. After adding the boilerplate support, I now see reports like these (which I haven't tried to decode or interpret yet):

Left wheel tilt:
```
ffff918e080d9300 773572175 C Ii:9:015:1 0:1 8 = 20000000 00000000
ffff918e080d9300 773572191 S Ii:9:015:1 -115:1 8 <
```
Right wheel tilt:
```
ffff918e080d9300 832888498 C Ii:9:015:1 0:1 8 = 40000000 00000000
ffff918e080d9300 832888511 S Ii:9:015:1 -115:1 8 <
```
So I assume that openrazer is sending some initialization command that opens up more complete button reporting. So far so good.

However, when I run "evtest /dev/input/by-id/usb-Razer_Razer_Basilisk_V2-event-mouse" I don't get any REL_HWHEEL events.

Is using the tilt wheel for REL_HWHEEL events supported for any mouse in openrazer? If so, can someone point me to how it's implemented for one of those mice, so I can get my bearings with implementing it for this one?

malsyned, 2020-05-05:
I dug through the driver and didn't find any support for the "Scroll Left" and "Scroll Right" mouse button events, so I added it to razer_raw_event. Something causes the mouse to sometimes stop sending those buttons, though, and I haven't figured out what the conditions are or what the fix is, so it's not ready for a pull request.

malsyned, 2020-05-05:
I see there's support for tilt wheels on other Razer mice, but the Basilisk V2 seems to work differently. Instead of sending a report on one of the keyboard interfaces, it sends one on the mouse interface, but only in driver mode (device_mode == 3:0). When it is in driver mode, bits 5 and 6 of the first byte of the report are used for Scroll Left and Scroll Right. The report descriptor defines those bits as padding, so I doubt the Linux HID layer even looks at them, which explains why they're being dropped on the floor.

For other mice, the tilt wheel events come in via the keyboard interface and get translated in razer_raw_event to F15 and F16, which the daemon listens for and takes from there. I don't see an easy way in the driver to associate the mouse interface with the keyboard interface so the event can be moved from the mouse interface to the keyboard interface.

My instinct is to detect the wheel left and wheel right events in the kernel deriver and emit REL_HWHEEL events instead. It may not tie nicely into openrazer-daemon, but at least it will make those buttons useful to userspace somehow. How does this sound?

AmadeusGraves, 2020-05-15:
The merge of the support is not done!
Issue: #1142

malsyned, 2020-05-26:
[razer-basilisk-v2-pcap.zip](https://github.com/openrazer/openrazer/files/4682770/razer-basilisk-v2-pcap.zip)
Attaching a bunch of pcapng files and a description of how they were created

KaeTuuN, 2021-01-29:
Any information here? Can I provide some Input?

NoXPhasma, 2021-05-18:
I would like to know as well if we can provide any more information on helping to expand the support for this mouse. The current state is pretty rudimentary.