openrazer/openrazer #1086: Manjaro terrible user experience and Razer Ornata Chroma not supported

trialism, 2020-04-08:
I installed this program for a new linux user and we needed to go through these:

`sudo useradd -G plugdev $(whoami)` - didn't work, but it was the first recommendation for the issue

`sudo gpasswd -a $USER plugdev` - after a restart, the daemon finally started, BUT

now it says "Device Not Recognised".

I also have manjaro and I get this:

`ModuleNotFoundError: No module named 'openrazer_daemon'`

This is what I used: `pacaur -S openrazer-meta`

trialism, 2020-04-08:
The installation page only mentions pacaur -S openrazer-meta, but now I know that we need the plgudev and linux-headers "hacks" - but those are still not enough.

trialism, 2020-04-08:
xyz.z3ntu.razergenie:

`
Gtk-Message: 12:52:28.776: Failed to load module "canberra-gtk-module"
Gtk-Message: 12:52:28.777: Failed to load module "canberra-gtk-module"
Qt: Session management error: Authentication Rejected, reason : None of the authentication protocols specified are supported and host-based authentication failed
Translation loaded: false
`

And after 30 seconds, it says "The OpenRazer daemon is not installed".

lah7, 2020-04-08:
OpenRazer 2.x's drawback is the DKMS driver and comes with the potential of things of different things going wrong. A lot of this will be a thing of the past once OpenRazer 3.0 handles everything in userspace.

Linux headers are required for DKMS to build them - but it's not as simple at a packaging level as users have different kernels (e.g. on Arch/Manjaro - `linux-headers` vs `linux-lts-headers`?)

> `sudo gpasswd -a $USER plugdev`

A re-login is required when changing [groups](https://wiki.archlinux.org/index.php/Users_and_groups#Group_management) for a logged in user, or a restart.

> `sudo useradd -G plugdev $(whoami)` - didn't work, but it was the first recommendation for the issue

Where's this please? `gpasswd` would be better advice since `useradd` attempts to create a new user.

> now it says "Device Not Recognised".

You must be using Polychromatic v0.3.12? I've got some improvements in the works for that project to improve the feedback when things ain't working. For now, [the troubleshooting page](https://github.com/openrazer/openrazer/wiki/Troubleshooting) has details to fix - though lately I've been thinking of improving it.

> `ModuleNotFoundError: No module named 'openrazer_daemon'`

What's saying this? Are you able to run `openrazer-daemon -Fv`?

If you haven't updated your AUR packages since the move to Python 3.7, you'll need to rebuild all your Python packages that came from the AUR.

    pacman -Qo /usr/lib/python3.7/site-packages/

trialism, 2020-04-08:
Relogin didn't work with gpasswd, only restart.
The useradd was in the bugtracker, I was searching for fixes.
We are using 0.3.12
openrazer-daemon -Fv starts, but it can't find the openrazer_daemon python module
I reinstalled a lot of python packages with pacaur.

trialism, 2020-04-08:
Alright, I reinstalled every python AUR package and openrazer works now, but for my friend the device is still not recognised, the logs don't tell much.

z3ntu, 2020-04-08:
Does `sudo modprobe razerkbd` work?

Also just a quick note: If you use the AUR you should be able to solve basic packaging "problems" like rebuilding Python packages once a new version of Python is released. If you can't do that then either you shouldn't use packages from the AUR or you should use another distro.

trialism, 2020-04-08:
A user shouldn't have to solve "problems" like this - such problems shouldn't exist at all because they're architectural errors. And python shouldn't be used for app development either because it is a terrible language with terrible package management practices.
And linux/amd users can't use non-rolling-release distros thanks to mesa packaging. Linux is just not user-friendly.
I use nvidia so I have no problems with using ubuntu as my main distribution but I had to provide support for this poor new linux/amd user.

trialism, 2020-04-08:
@lah7 we reinstalled everything and the (daemon) logs still say very little about the issue: https://pastebin.com/ry7yxS3V
polychromatic only says "Device Not Recognised"

lah7, 2020-04-08:
Please give us the details - what Razer device isn't being recognised? What's the output of this command?

> `sudo modprobe razerkbd`

d7eeem, 2020-04-17:
i have the same problem when running 
> > `sudo modprobe razerkbd`
 i get this 
`modprobe: FATAL: Module razerkbd not found in directory /lib/modules/5.4.31-1-MANJARO
`
using deathadder chroma

z3ntu, 2020-04-17:
@d7eeem Make sure your kernel headers are installed (in your case `linux54-headers`)

z3ntu, 2021-03-21:
I don't think this issue is relevant anymore, closing.

closed 2021-03-21 by z3ntu