openrazer/openrazer #1088: Ripple Support for Orbweaver Chroma (and some other stuff)

chabad360, 2020-04-08:
Can the ripple effect be added for the Orbweaver Chroma?

Plus, can the profile LEDs be remapped as they all hit the wrong stuff:

Current:

```
profile_led_green -> blue led
profile_led_red     -> green led
profile_led_blue    -> yellow led
```

Should be:

```
profile_led_green -> greed led
profile_led_red     -> yellow led
profile_led_blue    -> blue led
```

Thanks in advance!

z3ntu, 2020-04-19:
Please try applying the following patch to the daemon:
```diff
diff --git a/daemon/openrazer_daemon/hardware/keyboards.py b/daemon/openrazer_daemon/hardware/keyboards.py
index e6630fb..de1b046 100644
--- a/daemon/openrazer_daemon/hardware/keyboards.py
+++ b/daemon/openrazer_daemon/hardware/keyboards.py
@@ -265,7 +265,7 @@ class RazerOrbweaver(_RazerDeviceBrightnessSuspend):
         # self.key_manager.close()
 
 
-class RazerOrbweaverChroma(_RazerDeviceBrightnessSuspend):
+class RazerOrbweaverChroma(_RippleKeyboard):
     """
     Class for the Razer Orbweaver Chroma
     """
@@ -285,7 +285,8 @@ class RazerOrbweaverChroma(_RazerDeviceBrightnessSuspend):
                'keypad_get_profile_led_green', 'keypad_set_profile_led_green',
                'keypad_get_profile_led_blue', 'keypad_set_profile_led_blue',
                'get_macros', 'delete_macro', 'add_macro',
-               'keypad_get_mode_modifier', 'keypad_set_mode_modifier']
+               'keypad_get_mode_modifier', 'keypad_set_mode_modifier',
+               'set_ripple_effect', 'set_ripple_effect_random_colour']
 
     DEVICE_IMAGE = "https://assets.razerzone.com/eeimages/support/products/607/607_orbweaver_chroma.png"
 

```

chabad360, 2020-04-19:
@z3ntu I'll try it, but I'm quite confident it won't work as the ripple effect implementation is missing entirely for the Orbweaver Chroma.

https://github.com/openrazer/openrazer/blob/bd71e769d9239fc4ffac69c04cf3cc88b12d7bda/daemon/openrazer_daemon/misc/key_event_management.py#L787-L789

chabad360, 2020-04-19:
Can confirm, that didn't work, although it did add the ripple capability to the list (as expected), it also broke using custom effects (both through DBus and directly). I'm assuming that because it's done in software, the ripple system takes control of the frame and will overwrite any changes to it automatically. If there was a way (at least through DBus) to alert the daemon, that I now want to do something else, that would be great.

PR #1109 (2020-05-02): Add ripple effect for Orbweaver Chroma
This PR adds ripple effect support and closes #1088.
closed 2020-05-02 by z3ntu