openrazer/openrazer #1089: Profile LEDs not correct for Orbweaver Chroma

chabad360, 2020-04-08:
The profile LEDs for the Orbweaver Chroma should be remapped as they all hit the wrong stuff:

Currently:

```
profile_led_green -> blue led
profile_led_red   -> green led
profile_led_blue  -> yellow led
```

Should be:

```
profile_led_green -> greed led
profile_led_red   -> yellow led
profile_led_blue  -> blue led
```

I've identified the culprit: 

https://github.com/openrazer/openrazer/blob/3592f8197dc2bd143f02a5f996ba8f6feb9b7f80/driver/razercommon.h#L52-L54

As these are hardcoded (and because this was originally written for the Tatarus), there's doesn't seem to be a super easy way out.

lah7, 2020-04-09:
I can also confirm this is the case with **Tartarus Chroma**.

Looking at `git blame`, it [appears it was the Nostromo](https://github.com/openrazer/openrazer/blame/3592f8197dc2bd143f02a5f996ba8f6feb9b7f80/driver/razercommon.h#L52-L54) that has them in a different order.

https://github.com/openrazer/openrazer/commit/86af64a8f3d11a13dd09be37a26069b1998855b4#diff-6f38cc63b8755079814e6aa5d740b081L47-L49

chabad360, 2020-04-14:
I could give this a shot, but the question is whether to just change it to fit the Tartarus and Orbweaver, or to add another set of `#define`s (which in it of it self brings a naming challenge) and the requisite functions to make this work.

z3ntu, 2020-04-19:
From the information I have, the defines are correct from name to number. I'd say "fixing" this in the driver is the best solution, so adding an if or a switch to the respective functions which changes the id for those devices.

chabad360, 2020-04-19:
@z3ntu I'll give it a shot (I am super unfamiliar with C/C++, so here goes!).

chabad360, 2020-04-19:
@z3ntu I'd give it a shot, but I'm extremely unfamiliar with C/C++, and a quick look at that code doesn't give a boost of confidence. So... maybe you or @lah7 could give it a shot?

lah7, 2020-04-20:
I'm not confident with C either, but I'll have a look. :eyes:

commit 9f787a34 (2020-05-01): Use different profile IDs for some keypads

Orbweaver Chroma and Tartarus Chroma use different IDs then the
Nostromo.

Addresses 86af64a8f3d11a13dd09be37a26069b1998855b4
Fixes #1089
PR #1108 (2020-05-01): Use different profile LED IDs for some keypads
Fixes #1089. This commit fixes the wrong keypad profile LEDs on the Tartarus Chroma and Orbweaver Chroma.

It would be useful to know if this alternate ID is needed for other keypads, or just the ones that were  originally added (in which case, [Orbweaver may need to be added](https://github.com/openrazer/openrazer/commit/86af64a8f3d11a13dd09be37a26069b1998855b4#diff-04c6e90faac2675aa89e2176d2eec7d8L72-L76))

#### Devices to be checked

- [x] ~~Nostromo~~ (OK - https://github.com/openrazer/openrazer/commit/86af64a8f3d11a13dd09be37a26069b1998855b4# [... cut]
closed 2020-05-01 by z3ntu