openrazer/openrazer #1090: Macro giving differents outputs

Ratatinator97, 2020-04-09:
Hello I'm currently on Arch : 5.6.3-arch1-1. 
Openrazer package installed from AUR: openrazer-daemon 2.7.0-1, openrazer-driver-dkms 2.7.0-1, python-openrazer 2.7.0-1.

I have a Razer Mamba tournament edition going very well and a BlackWidow Chroma v2.

I wanted to thank you all for doing this awesome stuff !

I configured macros on my keyboard: 
- I'm on a AZERTY version but with the macros when I register "." or "@" (not tested all the keys) I got a different macro "." -> "M" and "@" -> "2". This occurs only in the "macro mode". 

- After bypassing this issue ( for the "." and the "@" I did like I had a QWERTY ie: "@" = "shift" + "2") I tested the macro. ISSUE: for example my email address : asidiras.csi@gmail.com gives me sometimes : asidiras.csi@g,qil.co, OR asidiras.csi@gmail.co,.
I think there is a conflict between QWERTY format and AZERTY format under the hood !

I don't really needed a help but just wanted to share the issue with you :)

lah7, 2020-04-09:
I have a similar issue where playing macros would use the US keyboard layout :us: instead of GB :gb:. My workaround is to run a command at log on:

    setxkbmap -layout 'gb'

Perhaps this might work for you? (Changing `gb` to a code in `/usr/share/X11/xkb/symbols/`), e.g. `fr`.

Ratatinator97, 2020-04-11:
> I have a similar issue where playing macros would use the US keyboard layout us instead of GB gb. My workaround is adding this command to run when I log on:
> 
> ```
> setxkbmap -layout 'gb'
> ```
> 
> Perhaps this might work for you? (Changing `gb` to a code in `/usr/share/X11/xkb/symbols/`), e.g. `fr`.

That did the trick ! Thanks :)

closed 2020-04-11 by Ratatinator97