openrazer/openrazer #1092: Razer BlackWidow Elite - Brightness not working

Severus157, 2020-04-10:
### Distribution

OS: Arch Linux x86_64 
Host: AX370-Gaming 3 
Kernel: 5.6.3-arch1-1 
Packages: 1361 (pacman) 
Shell: bash 5.0.16 
Resolution: 3440x1440 
DE: Plasma 
WM: KWin 
Terminal: konsole 
CPU: AMD Ryzen 5 2600 (12) @ 3.163GHz 
GPU: AMD ATI Radeon RX 5600 OEM/5600 XT / 5700/5700 XT 
Memory: 1972MiB / 32125MiB

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi,
I've got a bit of a problem with the Backgroundlighting of my BlackWidow Elite Keyboard. I have updated the System to a newer Kernel and now the Brightness is set to 0 from RazerGenie and I cannot change it.
The Keyboard itself seems fine, since the lighting works on a different PC not running Linux.

I've tried to add my user to plugdev group like some solutions proposed, but it didn't change anything.
The error I receive after starting up RazerGenie the second time is:

`● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2020-04-10 16:36:46 CEST; 1min 43s ago
       Docs: man:openrazer-daemon(8)
   Main PID: 857 (openrazer-daemo)
     CGroup: /user.slice/user-1000.slice/user@1000.service/openrazer-daemon.service
             └─857 openrazer-daemon

Apr 10 16:38:15 antergos-pc openrazer-daemon[857]: 2020-04-10 16:38:15 | razer.device0                  | DEBUG    | DBus call set_brightness
Apr 10 16:38:20 antergos-pc openrazer-daemon[857]: 2020-04-10 16:38:20 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowElite:IO1839F31107680, 'setBrightness', 178)
Apr 10 16:38:20 antergos-pc openrazer-daemon[857]: 2020-04-10 16:38:20 | razer.device1                  | DEBUG    | Got observer message: ('effect', RazerBlackWidowElite:IO1839F31107680, 'setBrightness', 178)
Apr 10 16:38:20 antergos-pc openrazer-daemon[857]: 2020-04-10 16:38:20 | razer.device0                  | DEBUG    | DBus call set_brightness
Apr 10 16:38:25 antergos-pc openrazer-daemon[857]: 2020-04-10 16:38:25 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowElite:IO1839F31107680, 'setBrightness', 204)
Apr 10 16:38:25 antergos-pc openrazer-daemon[857]: 2020-04-10 16:38:25 | razer.device1                  | DEBUG    | Got observer message: ('effect', RazerBlackWidowElite:IO1839F31107680, 'setBrightness', 204)
Apr 10 16:38:25 antergos-pc openrazer-daemon[857]: 2020-04-10 16:38:25 | razer.device0                  | DEBUG    | DBus call set_brightness
Apr 10 16:38:30 antergos-pc openrazer-daemon[857]: 2020-04-10 16:38:30 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowElite:IO1839F31107680, 'setBrightness', 229)
Apr 10 16:38:30 antergos-pc openrazer-daemon[857]: 2020-04-10 16:38:30 | razer.device1                  | DEBUG    | Got observer message: ('effect', RazerBlackWidowElite:IO1839F31107680, 'setBrightness', 229)
Apr 10 16:38:30 antergos-pc openrazer-daemon[857]: 2020-04-10 16:38:30 | razer.device0                  | DEBUG    | DBus call set_brightness`

z3ntu, 2020-04-19:
Can you check that the effect isn't set to "Off" (or "None"), because if that effect is active, the brightness won't matter.

Severus157, 2020-04-26:
Hi z3ntu,
The brightness is always recognised as off, even if it is on. Tried with Polychromatic as well as Razergenie.
Also the brightness is completly uncontrollable. It seems like it cannot be recognised, which setting is true.

Also it looks like the SerialNr cannot be recognised as well.
[image]

lsusb gives out pretty strange things for the Razer Devices on the new Kernels. Problem persists on 5.6.6.
```
[s@antergos-pc ~]$ sudo lsusb -v | grep "1532"
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Bus 003 Device 005: ID 1532:005c Razer USA, Ltd 802.11n WLAN Adapter
  idVendor           0x1532 Razer USA, Ltd
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Bus 003 Device 004: ID 1532:0228 Razer USA, Ltd 
  idVendor           0x1532 Razer USA, Ltd
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
cannot read device status, Resource temporarily unavailable (11)
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
```
If I can get you any other outputs let me know.

Severus157, 2020-05-20:
Still the same problem. Brightness completly crashes after sleep mode also. And before cannot be set anyways.

Solution steps I tried together with the new Kernel Update. Of completly removing openrazer and reinstall. Still nothing.
Have tried Polychromatic instead of RazerGenie. Error there: Daemon not Responding
`org.freedesktop.DBus.Error.ServiceUnknown: The name :1.114 was not provided by any .service files`

closed 2020-12-25 by stale[bot]