openrazer/openrazer #1096: I really don't know what's happening

matiusX, 2020-04-13:
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-04-12 21:54:10 -04; 24s ago
     Docs: man:openrazer-daemon(8)
  Process: 2960 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 2960 (code=exited, status=1/FAILURE)

Apr 12 21:54:10 matheus-Aspire-VX5-591G systemd[1573]: Starting Daemon to manage razer devices in userspace...
Apr 12 21:54:10 matheus-Aspire-VX5-591G openrazer-daemon[2960]: Traceback (most recent call last):
Apr 12 21:54:10 matheus-Aspire-VX5-591G openrazer-daemon[2960]:   File "/usr/bin/openrazer-daemon", line 11, in <module>
Apr 12 21:54:10 matheus-Aspire-VX5-591G openrazer-daemon[2960]:     from openrazer_daemon.daemon import RazerDaemon, __version__
Apr 12 21:54:10 matheus-Aspire-VX5-591G openrazer-daemon[2960]:   File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 15, in <module>
Apr 12 21:54:10 matheus-Aspire-VX5-591G openrazer-daemon[2960]:     import setproctitle
Apr 12 21:54:10 matheus-Aspire-VX5-591G openrazer-daemon[2960]: ModuleNotFoundError: No module named 'setproctitle'
Apr 12 21:54:10 matheus-Aspire-VX5-591G systemd[1573]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Apr 12 21:54:10 matheus-Aspire-VX5-591G systemd[1573]: openrazer-daemon.service: Failed with result 'exit-code'.
Apr 12 21:54:10 matheus-Aspire-VX5-591G systemd[1573]: Failed to start Daemon to manage razer devices in userspace.

lah7, 2020-04-13:
> ModuleNotFoundError: No module named 'setproctitle'

Not all the dependencies are installed, so the daemon won't start.

What distro are you using? Try installing `python-setproctitle` or on Ubuntu/Debian-based `python3-setproctitle`.

z3ntu, 2020-04-13:
In the case you're running Arch Linux you should rebuild all packages printed by the command `pacman -Qo /usr/lib/python3.7`

matiusX, 2020-04-13:
I'm using ubuntu 18.04.4 LTS

z3ntu, 2020-04-13:
Can you check if the `python3-setproctitle` package is installed?

matiusX, 2020-04-13:
setproctitle-1.1.10

z3ntu, 2020-04-13:
Does the file `/usr/lib/python3/dist-packages/setproctitle.cpython-36m-x86_64-linux-gnu.so` exist in the file system (potentially with a slightly different name)?

matiusX, 2020-04-13:
I've tried to acess but, bash: cd: /usr/lib/python3/dist-packages/setproctitle.cpython-36m-x86_64-linux-gnu.so: is not a directory

z3ntu, 2020-04-13:
`ls /usr/lib/python3/dist-packages/setproctitle.cpython-36m-x86_64-linux-gnu.so` ?

matiusX, 2020-04-13:
output: /usr/lib/python3/dist-packages/setproctitle.cpython-36m-x86_64-linux-gnu.so

z3ntu, 2020-04-13:
So the file does exist.. Did you compile python from source or something like this on this machine?

matiusX, 2020-04-13:
To install python, i just use 'sudo apt install python3.7'

z3ntu, 2020-04-14:
What's the output of the command `python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())"` ?

matiusX, 2020-04-14:
/usr/lib/python3/dist-packages

z3ntu, 2020-04-14:
And `python3 -c 'import setproctitle'` works fine (exits without an error)?

matiusX, 2020-04-14:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'setproctitle'

I don't know why, because setproctitle is installed

lah7, 2020-04-15:
I think I see the problem - you've installed `python3.7` but `python3.6` is the default version for Ubuntu 18.04, which is what packages like `python3-setproctitle` and OpenRazer will install to.

Just to be sure, if you run:

    echo $PATH

It should output this on Ubuntu 18.04:

    /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

And this command:

    python3 -c "import sys; print(sys.path)"

Should output:

    ['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages']

So, you may just need to:

    sudo apt autoremove python3.7

It would be more reliable to stick with the version default to your installation, unless you have a specific reason to need `python3.7` (in which there are workarounds).

    $ python3 --version
    Python 3.6.9

matiusX, 2020-05-04:
that works well, thanks!!

closed 2020-05-04 by matiusX