openrazer/openrazer #1099: Ripple effect not working on Blackwidow 2019

mcat95, 2020-04-15:
### Distribution

Arch Linux

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

When I try to set the effect to the 'ripple' effect on my Razer Blackwidow 2019 (ID 1532:0241), all leds turn off and stay that way. I have tried restarting several times, both my system and the daemon. All other effects work perfectly.

Here is the python script I'm using to test this: 
`python -c "import openrazer.client; a=openrazer.client.DeviceManager().devices[0]; print(a.name); a.fx.ripple(0,255,0)"`

Here is the daemon log when the script is executed:
```
2020-04-15 23:49:26 | razer                          | DEBUG    | DBus called get_serial_list
2020-04-15 23:49:26 | razer.device0                  | DEBUG    | DBus call get_device_type
2020-04-15 23:49:26 | razer.device0                  | DEBUG    | DBus call get_device_name
2020-04-15 23:49:26 | razer.device0                  | DEBUG    | DBus call get_device_type
2020-04-15 23:49:26 | razer.device0                  | DEBUG    | DBus call get_firmware
2020-04-15 23:49:26 | razer.device0                  | DEBUG    | DBus call version
2020-04-15 23:49:26 | razer.device0                  | DEBUG    | DBus call has_matrix
2020-04-15 23:49:26 | razer.device0                  | DEBUG    | DBus call has_matrix
2020-04-15 23:49:26 | razer.device0                  | DEBUG    | DBus call get_keyboard_layout
2020-04-15 23:49:26 | razer.device0                  | DEBUG    | DBus call set_ripple_effect
2020-04-15 23:49:26 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidow2019:IO2009F32805049, 'setRipple', dbus.Byte(0), dbus.Byte(255), dbus.Byte(0), dbus.Double(0.05))
```

Toasterbirb, 2020-04-19:
I'm also experiencing this same issue with the GUI apps provided in the README file.

z3ntu, 2020-04-19:
On my BlackWidow Chroma the ripple effect still works properly.

The line https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/keyboards.py#L907 for the BlackWidow 2019 looks wrong (compared to the /dev/input/by-id output in #844).

Can you try changing that line by removing `_2019` and restarting the daemon? You can edit the file in-place in `/usr/lib/python3.*/site-packages/openrazer_daemon/hardware/keyboards.py` (or `/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py` on Debian-based distros)

mcat95, 2020-04-19:
Yep! That seems to fix the problem.

z3ntu, 2020-04-24:
Thanks for testing!

commit ee5aa774 (2020-04-24): Fix EVENT_FILE_REGEX for BlackWidow 2019

Fixes the ripple effect on that device.

Fixes #1099
closed 2020-04-24 by z3ntu