openrazer/openrazer #1104: Polychromatic not identifying Razer Abyssus Essential

T3CHKOMMIE, 2020-04-18:
Hello,

I have a Razer Abyssus Essential that is not being picked up by the daemon. I have enabled modprobe rasermouse restarted daemon and bounced the mouse a few times. Still no joy

Legion:~$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 002: ID 17ef:3069 Lenovo ThinkPad TBT3 LAN
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 004: ID 1532:0c01 Razer USA, Ltd Razer Goliathus Chroma
**Bus 007 Device 003: ID 1532:006b Razer USA, Ltd Razer Abyssus Essential**
Bus 007 Device 002: ID 1532:022a Razer USA, Ltd Razer Cynosa Chroma
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 004: ID 17ef:3066 Lenovo ThinkPad Thunderbolt 3 Dock MCU
Bus 005 Device 003: ID 05ac:1262 Apple, Inc. iPod Nano 3.Gen
Bus 005 Device 002: ID 17ef:306a Lenovo ThinkPad Thunderbolt 3 Dock USB Audio
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 13d3:56a6 IMC Networks Integrated Camera
Bus 001 Device 003: ID 048d:c935 Integrated Technology Express, Inc. Corsair Gaming K95 RGB PLATINUM Keyboard
Bus 001 Device 002: ID 048d:c936 Integrated Technology Express, Inc. ITE Device(8910)
Bus 001 Device 005: ID 8087:0aaa Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Legion:~$ lsmod | grep razer
razeraccessory         40960  0
razercore              32768  0
razerkraken            20480  0
**razermouse             81920  0**
razerkbd               65536  0
razermousemat          36864  0
hid                   131072  10 i2c_hid,razercore,usbhid,hid_multitouch,razerkraken,razerkbd,hid_generic,razeraccessory,razermousemat,razermouse

Mouse pad (goliathus) and keyboard are discoverable and configurable.

z3ntu, 2020-04-19:
Hi, the Abyssus Essential isn't included in a stable release yet, so you have to install nightly builds (daily PPA on Ubuntu or openrazer-git on Arch) until we get a new release out.

Psychotechnopath, 2020-04-19:
I'm experiencing the same issue. Where do I find such a nightly build, and are you planning on releasing a version with support for the Abyssus Essential?

z3ntu, 2020-04-24:
@Psychotechnopath I'm waiting for a final feature to get ready (PR is open for that) but it's taking a bit longer than expected. But a new release should be done in the next couple of weeks.

Psychotechnopath, 2020-04-24:
@z3ntu Great, Thx for the reply =)

nightsky30, 2020-05-03:
This should be good now :)

z3ntu, 2020-05-03:
Yep, 2.8.0 is released!

closed 2020-05-03 by z3ntu

Kern3l-Panic, 2020-05-03:
Thank you! I've been waiting for these drivers for some time now. Glad to see they work too. One thing to add: I see in the tray applet that the breathing effect is available, but not in the controller. Again, thank you.

lah7, 2020-05-04:
Polychromatic's work-in-progress v1.0.0 actually needs #888 ([`effect_storage`](https://github.com/openrazer/openrazer/tree/effect_storage)) for effects to appear in the Controller. We hope to have this branch ready in the upcoming weeks.

tildearrow, 2020-05-04:
@lah7 That will not be worked on. It's closed because v2.8.0 was released.