openrazer/openrazer #1107: Polychromatic Razer (Ubuntu 20.04)

style2k6, 2020-04-21:
Hello,

I installed openRazer with Polychromatic, but i get the message that one device is not recoganized.

log of lusb:

Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

Bus 005 Device 003: ID 8087:0025 Intel Corp.

Bus 005 Device 002: ID 1462:7b85 Micro Star International PRO CARBON

Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

Bus 001 Device 027: ID 1532:002e Razer USA, Ltd RZ01-0058 Gaming Mouse [Naga]

Bus 001 Device 026: ID 1532:021e Razer USA, Ltd

Bus 001 Device 025: ID 1532:0c01 Razer USA, Ltd

Bus 001 Device 024: ID 05e3:0610 Genesys Logic, Inc. 4-port hub

Bus 001 Device 002: ID 2516:0051 Cooler Master Co., Ltd. AMD SR4 lamplight Control

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

im newbie of linux, so need help. Thanks

z3ntu, 2020-04-21:
Hi, what's the output of the command  `sudo dkms install openrazer-driver/2.7.0`  ?

style2k6, 2020-04-21:
thank you for your answer,here the output:
Error! Your kernel headers for kernel 5.4.0-25-lowlatency cannot be found.
Please install the linux-headers-5.4.0-25-lowlatency package,
or use the --kernelsourcedir option to tell DKMS where it's located

lah7, 2020-04-21:
Hi, you'll need to install the headers for your kernel, which will allow you to build the driver. DKMS will keep it updated automatically whenever you update the kernel.

These commands should get you up and running:

    sudo apt install linux-headers-lowlatency
    sudo apt install --reinstall openrazer-driver-dkms
    sudo gpasswd -a $USER plugdev

Then reboot the computer :ok_hand:

style2k6, 2020-04-21:
thank you Lah7 , have done what you wrote, but still the same failure

lah7, 2020-04-21:
Okay, what is the output of these commands?

    sudo dkms install openrazer-driver/2.7.0
    sudo modprobe razerkbd

style2k6, 2020-04-21:
Error! Your kernel headers for kernel 5.4.0-25-lowlatency cannot be found.
Please install the linux-headers-5.4.0-25-lowlatency package,
or use the --kernelsourcedir option to tell DKMS where it's located

modprobe: FATAL: Module razerkbd not found in directory /lib/modules/5.4.0-25-lowlatency

lah7, 2020-04-21:
Hmm, odd. The `linux-headers-lowlatency` package should have installed the relevant package. You can explicitly install the package for the kernel you have like so:

    sudo apt install linux-headers-5.4.0-25-lowlatency

Then repeat the commands above, and reboot if things were successful.

---

Ubuntu 20.04 is still in beta until Thursday - the latest kernel is `5.4.0.26.32` - you may want to update the rest of your system using **Software Updater**.

style2k6, 2020-04-21:
no its working, polychromatic is working! thank you so much!

closed 2020-04-21 by lah7