openrazer/openrazer #1110: Driver Installation & Binding Help

BounceyQueso, 2020-04-22:
### Distribution

Debian v10.0

### Support

- [ X ] Have you tried rebooting?
Yes, I tried rebooting.
- [ X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Yes, I have worked through a few similar paths with no success.

When I run: "modprobe razerkbd"

I get the following error: "modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.19.57-v7l+"

I am able to run "sudo apt-get update" and it re-checks I have openrazer installed. I'm not sure where to go next. 

Should be noted I am trying to connect the following to Raspberry Pi 4 w/ Debian 10.0
- Mamba Elite mouse
- Ornata Chroma

z3ntu, 2020-04-23:
You have to install the kernel headers for your kernel. The package is normally called `linux-headers-*` but I'm not sure of the exact name in your case.

Can you run the command `dpkg -l | grep linux` and post the output here?

BounceyQueso, 2020-04-23:
When I run that command I get the following:
dpkg -l | grep linux
ii  binutils-arm-linux-gnueabihf          2.31.1-16+rpi1                        armhf        GNU binary utilities, for arm-linux-gnueabihf target
ii  console-setup-linux                   1.191                                 all          Linux specific part of console-setup
ii  fonts-linuxlibertine                  5.3.0-4                               all          Linux Libertine family of fonts
ii  libselinux1:armhf                     2.8-1+b1                              armhf        SELinux runtime shared libraries
ii  libv4l-0:armhf                        1.16.3-3                              armhf        Collection of video4linux support libraries
ii  libv4lconvert0:armhf                  1.16.3-3                              armhf        Video4linux frame format conversion library
ii  linux-base                            4.6                                   all          Linux image base package
ii  linux-libc-dev:armhf                  4.18.20-2+rpi1                        armhf        Linux support headers for userspace development
ii  util-linux                            2.33.1-0.1                            armhf        miscellaneous system utilities
ii  v4l-utils                             1.16.3-3                              armhf        Collection of command line video4linux utilities

Also, I tried to run "sudo apt-get install raspberrypi-kernel-headers" and it completed with out upgrading/installing because they already were installed.

BounceyQueso, 2020-04-23:
When I run: sudo dkms install openrazer-driver/2.7.0 I get the following:

sudo dkms install openrazer-driver/2.7.0
Error! echo
Your kernel headers for kernel 4.19.57-v7l+ cannot be found at
/lib/modules/4.19.57-v7l+/build or /lib/modules/4.19.57-v7l+/source.

lah7, 2020-04-23:
I just tried Raspbian 10 (Buster) 2020-02 image on a fresh Raspberry Pi 4 and can confirm it does work on the hardware.

Your Pi seems to be running an older kernel (`4.19.57`) - mine is `4.19.97`. It's possible `raspberrypi-kernel-headers` installed the headers for a newer/mismatching kernel, and your `raspberrypi-kernel` package needs updating to match:

    sudo apt upgrade raspberrypi-kernel
    sudo reboot

Unfortunately, Raspbian doesn't provide separate packages for each version of the kernel unlike the desktop version.

BounceyQueso, 2020-04-23:
[SOLVED]

Thanks for the help, here is what I did.
1. Fresh install of NOOBS
2. sudo apt upgrade respberrypi-kernel; Followed by sudo reboot
3. Installed the repo & built the packages. Then installed openrazer
4. Resolved install issues with apt --fix-broken install
5. Followed wiki for the driver guide to bind drivers to mouse and keyboard
6. Installed polychromatic
7. sudo reboot
8. IT WORKS! :) YAY. Thanks again. Hopefully this will help others.

closed 2020-04-23 by BounceyQueso