openrazer/openrazer #1111: Razer Lancehead TE LEDs not working

t0msk, 2020-04-23:
### Distribution
Arch Linux - 5.6.2-arch1-2

openrazer-2.7.0-1

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

Log from `journalctl`:
```
Apr 23 10:32:15 tomsk-PC razer_mount[592]: Driver razermouse
Apr 23 10:32:15 tomsk-PC razer_mount[593]: Driver razermouse
Apr 23 10:32:15 tomsk-PC razer_mount[594]: Device_ID 0003:1532:0060.0005
Apr 23 10:32:15 tomsk-PC razer_mount[595]: Device_ID 0003:1532:0060.0007
Apr 23 10:32:15 tomsk-PC razer_mount[596]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.0007/
Apr 23 10:32:15 tomsk-PC razer_mount[597]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.0005/
Apr 23 10:32:15 tomsk-PC razer_mount[600]: Driver razermouse
Apr 23 10:32:15 tomsk-PC razer_mount[606]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0060.0005/
Apr 23 10:32:15 tomsk-PC razer_mount[604]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0060.0007/
Apr 23 10:32:15 tomsk-PC razer_mount[607]: Device_ID 0003:1532:0060.0006
Apr 23 10:32:15 tomsk-PC razer_mount[610]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0060.0006/
Apr 23 10:32:15 tomsk-PC razer_mount[612]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0060.0006/
```
Log from `~/.local/share/openrazer/logs/razer.log`:
```
2020-04-23 10:32:45 | razer                          | INFO     | Initialising Daemon (v2.7.0). Pid: 802
2020-04-23 10:32:45 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Base ---------------------- (1532:023B)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Mercury ------------------- (1532:0240)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Adv ----------------------- (1532:023A)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Base ---------------------- (1532:0246)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeMid2019Mercury ---------------- (1532:0245)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth2019 ------------------- (1532:0239)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2019 --------------- (1532:024A)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerCynosaChroma ----------------------- (1532:022A)
2020-04-23 10:32:45 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdder1800 --------------------- (1532:0038)
[... 109 lines cut ...]
2020-04-23 10:32:45 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2020-04-23 10:32:45 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2020-04-23 10:32:45 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2020-04-23 10:32:45 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2020-04-23 10:32:45 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2020-04-23 10:32:45 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
[... cut ...]
2020-04-23 10:32:47 | razer.device0                  | DEBUG    | DBus call get_logo_brightness
2020-04-23 10:32:47 | razer.device0                  | DEBUG    | DBus call get_scroll_brightness
```

I don't know what happened, but suddenly my LEDs (all of them) are not working (they are off) anymore on my Razer Lancehead TE, but my mouse is working correctly only LEDs are off.
For frontend app I use `Polychromatic` and `RazerGenie` I tried to change effect and color of LEDs, but it didn't help, I tried both apps. My mouse is pretty new, it is only 4 months old.

lah7, 2020-04-23:
It looks like the driver and daemon is working normally - have you tried changing the brightness?

I have a feeling that in `~/.config/openrazer/razer.conf`, you may have `devices_off_on_screensaver` set to `True`. It might be that the screensaver came on (or the system went to sleep) and upon unlocking, the brightness wasn't restored properly after unlock/resume. Try changing this to `False`.

Both Polychromatic 0.3.12 and RazerGenie 0.8.1 only support the logo/scroll LEDs at the moment. 

You can try setting them all to 100% brightness and a green static using the daemon directly by running this:
```
python3 -c "import openrazer.client; a = openrazer.client.DeviceManager().devices[0]; a.fx.misc.left.static(0,255,0); a.fx.misc.left.brightness = 100; a.fx.misc.right.static(0,255,0); a.fx.misc.right.brightness = 100; a.fx.misc.logo.static(0,255,0); a.fx.misc.logo.brightness = 100; a.fx.misc.scroll_wheel.static(0,255,0); a.fx.misc.scroll_wheel.brightness = 100;"
```
(Assumes this is the only OpenRazer device)

t0msk, 2020-04-23:
Oh it is working :) Thank you :) Problem was that `devices_off_on_screensaver` was set to `True`, can I ask you one more question? Why I cannot change color of side LEDs in Polychromatic / RazerGenie? Looks like that in command line using your script I can :)

lah7, 2020-04-23:
You're welcome! :+1:

The driver/daemon supports it but the frontends still have a bit of catch up to do.

I develop **Polychromatic** - I see you've already left an issue a few months ago: https://github.com/polychromatic/polychromatic/issues/225 :slightly_smiling_face:  I do have these zones implemented in a [`dev-1.0.0`](https://github.com/polychromatic/polychromatic/tree/dev-1.0.0) branch but it's quite buggy at the moment (see https://github.com/polychromatic/polychromatic/issues/244#issuecomment-618086850). This version depends on #888 being merged. Progress has been slow, but not abandoned! :snail: 

**RazerGenie** has a pull request open to add support: https://github.com/z3ntu/RazerGenie/pull/54

@nightsky30 has tweaked both stable version of the applications in a fork which you could use in the meantime:

* https://github.com/nightsky30/RazerGenie/tree/side-leds (in a pull request)
* https://github.com/nightsky30/polychromatic/tree/side-leds

closed 2020-04-23 by lah7

t0msk, 2020-04-23:
Ah I see :) RazerGenies pull request is about 1 year old, I wonder why it is not merged, when will be `https://github.com/polychromatic/polychromatic/tree/dev-1.0.0` released? :)

lah7, 2020-04-23:
I cannot say, development is unpredictable! :microscope:   "As soon as it's ready", some point this year? :crossed_fingers: