openrazer/openrazer #1112: Device Not Recognised

gabrielfnd, 2020-04-24:
Hi,
My device is not recognized even though everything in troubleshooting guide seem to be all cleared.
Tried many things but couldn't sort it out.

`sudo dkms status`
`openrazer-driver, 2.7.0, 4.15.0-96-generic, x86_64: installed`

`sudo modprobe razerkbd` 
This one was throwing some ERROR, but after some searching, I disabled the secure boot and now it is **not** showing any errors or success

`lsusb | grep '1532'`
`Bus 001 Device 002: ID 1532:022c Razer USA, Ltd`

`openrazer-daemon -Fv`
`Starting daemon.`
`2020-04-24 14:46:23 | razer                          | INFO     | Initialising Daemon (v2.7.0). Pid: 16478`
`2020-04-24 14:46:23 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor`
`2020-04-24 14:46:23 | razer                          | DEBUG    | Loaded device specification: RazerTartarusChroma --------------------- (1532:0208)`
`2020-04-24 14:46:23 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus`
`2020-04-24 14:46:23 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus`
`2020-04-24 14:46:23 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus`
`2020-04-24 14:46:23 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus`
`2020-04-24 14:46:23 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus`
`2020-04-24 14:46:23 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus`
`2020-04-24 14:46:23 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus`
`2020-04-24 14:46:23 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus`
`2020-04-24 14:46:23 | razer                          | INFO     | Serving DBus`
`2020-04-24 14:46:40 | razer                          | DEBUG    | DBus called get_serial_list`

`grep 'plugdev' /etc/group`
plugdev:x:46:**gabriel**,root
Yes, I added root to the group just to try fixing the issue.

- Distribution
I'm on Ubuntu 18.04.4 LTS

- Support
[Yes] Have you tried rebooting?
[Yes] Have you looked at the Troubleshooting page?

My device is **Razer Cynosa Chroma Pro**. 
Even tho only its regular version is not listed, someone could still use the PRO version normally, as seen here https://www.bleepingcomputer.com/forums/t/706577/razer-cynosa-chroma-pro-keyboard-on-linux/
I am currently trying to use **Polychromatic**.

lah7, 2020-04-24:
Hi, the Pro model (`1532:022C`) isn't supported yet, just the non-Pro (`1532:022A`). There is an open issue at https://github.com/openrazer/openrazer/issues/464 in which will need adding to the driver.

I'm not sure how the guy on that forum is using the Pro, unless they modified the driver themselves (and didn't contribute their efforts :disappointed:) or has mistaken the Pro for the non-Pro model.

gabrielfnd, 2020-04-24:
Hey, @lah7.
Thanks for the reply!
Well, that’s unfortunate. Anyway, I will be monitoring the other issue

closed 2020-04-24 by gabrielfnd