openrazer/openrazer #1113: Razor Mamba Wireless (2012) not working properly...ish

TeamFahQ, 2020-04-25:
### Distribution

### Support

- [ X] Have you tried rebooting?
- [ X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

If I echo a number to any of:

`/sys/bus/hid/drivers/razermouse/0003:1532:0025.0001/scroll_led_brightness'
/sys/bus/hid/drivers/razermouse/0003:1532:0025.0001/scroll_led_effect'
/sys/bus/hid/drivers/razermouse/0003:1532:0025.0001/scroll_led_rgb'
/sys/bus/hid/drivers/razermouse/0003:1532:0025.0001/scroll_led_state'`

while having them open in text editor, I get the notificaiton that the file has change, but the number sent to the file, i.e. 1 for scroll_rgb_state, doesn't change. It stays at 0, and nothing happens. If I manually change the number to 1, again, nothing happens (because the device is on) but the number stays 1. Subsequently, when I change it to 0, the number stays and the mouse and base led's turn off.

This is the same for all listed files, except scroll_led_rgb. Regardless if I try echo command in terminal or manually write a number, nothing happens. But I am able to change colors with OpenRGB.

Furthermore, I can manually chang ethe effects (static, breathing, flashing, spectrum) manually, but echo has no effect and using OpenRGB causes it to get stuck in spectrum. As previously noted, OpenRGB does not set flashing or breathing mode, but instead, turns the device off, and I have to manually edit brightness and state in-order-to get the device working again.

According to openrazer, openrazer-daemon should be running, but this is not true. Regardless if it is running or not, everything mentioned above reacts the same.

When I start OpenRGB and switch to my mouse in devices tab, it reports the device is off. I'm guessing this is related to the fact the the number in the listed files always stays at 0.

I have also tried polychromatic, but the results are the same.

When enabling specturm mode, I have to completely uninstall openrazer and all it's related compnents, reboot, reinstall everything, add user to plugdev, reboot.

Also, battery always spams that it is 0.0% charged, but on the mouse it is 100%. Again, I think this may be related to the fact that all values in the driver directory are 0.

I am currently running Linux Mint 19.3. I use a custom kernel based off 5.3, and I have tried with stock kernels 5.3 and 5.0. But the reults are the same regardless of the kernel I use.

And one last note, when I use openrazer-daemon -vF, I get a lot of log spam (as noted) but there are never any errors, and after I changed to spectrum mode, the only thing it would log was brightness.

z3ntu, 2020-04-25:
Can you please explain your issue in one or two sentences? I've read through your message twice and I'm still not sure what's not working for you.

TeamFahQ, 2020-04-25:
Well, I thought the more info, the better picture you would have. Basically, things don't work right. Battery charge state is wrong. After selecting spectrum mode, it gets stuck in specturm mode and cannot change to static. The brightness and scroll wheel always says off.

z3ntu, 2020-05-01:
Can you get me the output of `dmesg | grep razermouse` ?

closed 2020-10-04 by stale[bot]