openrazer/openrazer #1115: "device not recognized" error on Polychromatic

legoman8304, 2020-04-26:
Hello all, 
I own a "Razer Ornata Chroma" (1532:021E) I am running Manjaro (5.5.19-1-MANJARO) with a Deepin 15 DE. I have tried all 3 listed frontend programs and all of them give me the same error. I am part of the plugdev group and all kernel headers are installed properly. I have also tried all the solutions at https://github.com/openrazer/openrazer/wiki/Troubleshooting 

Why am I getting this error and what can I do to fix it?

Any help would be greatly appreciated.

lah7, 2020-04-27:
Hello, what's the output of these commands?

    sudo modprobe razerkbd
    openrazer-daemon -Fv

It sounds like you may need to install the driver for DKMS again if you installed the kernel headers package after installing OpenRazer:

    sudo dkms install openrazer-driver/2.7.0

legoman8304, 2020-04-27:
So kind of an odd change of events but I booted up today to run hose commands and when I looked at my tray I saw that the daemon had recognized my keyboard and polychromatic now works. I'm not sure why, I'm not sure how, but it now works. Before I close this issue, I had a question: Can I have more than one frontend installed or can I only have one? Thanks

lah7, 2020-04-27:
Good to hear!

You can install multiple frontends if you like. Though, until #888 is merged, each frontend may report different things about the current status of the device.

closed 2020-04-27 by lah7