openrazer/openrazer #1116: AttributeError: 'RazerHuntsman' object has no attribute 'key_manager'

nkprince007, 2020-04-27:
```
● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Mon 2020-04-27 15:30:22 CDT; 40ms ago
       Docs: man:openrazer-daemon(8)
    Process: 200327 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=0/SUCCESS)
   Main PID: 200327 (code=exited, status=0/SUCCESS)

Apr 27 15:30:21 sakura openrazer-daemon[200327]:   File "/usr/lib/python3.8/site-packages/openrazer_daemon/hardware/device_base.py", line 431, in __del__
Apr 27 15:30:21 sakura openrazer-daemon[200327]:     self.close()
Apr 27 15:30:21 sakura openrazer-daemon[200327]:   File "/usr/lib/python3.8/site-packages/openrazer_daemon/hardware/device_base.py", line 345, in close
Apr 27 15:30:21 sakura openrazer-daemon[200327]:     self._close()
Apr 27 15:30:21 sakura openrazer-daemon[200327]:   File "/usr/lib/python3.8/site-packages/openrazer_daemon/hardware/keyboards.py", line 59, in _close
Apr 27 15:30:21 sakura openrazer-daemon[200327]:     super(_RippleKeyboard, self)._close()
Apr 27 15:30:21 sakura openrazer-daemon[200327]:   File "/usr/lib/python3.8/site-packages/openrazer_daemon/hardware/keyboards.py", line 43, in _close
Apr 27 15:30:21 sakura openrazer-daemon[200327]:     self.key_manager.close()
Apr 27 15:30:21 sakura openrazer-daemon[200327]: AttributeError: 'RazerHuntsman' object has no attribute 'key_manager'
Apr 27 15:30:22 sakura systemd[2559]: openrazer-daemon.service: Succeeded.
```

chabad360, 2020-04-27:
This may be caused by inserting and removing the device too quickly. Try restarting the daemon, run `systemctl --user restart openrazer-daemon.service`.
If that doesn't help, can you give the output of `journalctl --user -b0 -u openrazer-daemon`?

nkprince007, 2020-04-30:
It's not very frequent that this incident occurs, but it's a desktop and I usually don't unplug my keyboard. I'll post you the jounral logs once I see this issue again.

closed 2020-09-01 by nkprince007