openrazer/openrazer #1121: Razer Blade 2019 Advanced reported as Mercury 2019

philippgerard, 2020-04-28:
### Distribution

Manjaro 20, unstable, Kernel 5.6.7-1-MANJARO

### Device

Razer Blade 2019 Advanced (black, 240Hz HD display)

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

### Expected behavior

Correct device detected, settings for keyboard etc. applied

### Observed behavior

Razer Blade 2019 Mercury detected. OpenRazer allows only Lightning style to be set, also reports faulty model name.

### Logs

```2020-04-28 11:10:32 | razer                          | INFO     | Initialising Daemon (v2.7.0). Pid: 1371
2020-04-28 11:10:32 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Base ---------------------- (1532:023B)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Mercury ------------------- (1532:0240)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Adv ----------------------- (1532:023A)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Base ---------------------- (1532:0246)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeMid2019Mercury ---------------- (1532:0245)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth2019 ------------------- (1532:0239)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2019 --------------- (1532:024A)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerCynosaChroma ----------------------- (1532:022A)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdder1800 --------------------- (1532:0038)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdder2013 --------------------- (1532:0037)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdder3500 --------------------- (1532:0054)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdder3_5G --------------------- (1532:0016)
2020-04-28 11:10:32 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
[... 70 lines cut ...]
2020-04-28 11:10:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightRandom method to DBus
2020-04-28 11:10:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightSingle method to DBus
2020-04-28 11:10:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightDual method to DBus
2020-04-28 11:10:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRipple method to DBus
2020-04-28 11:10:32 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRippleRandomColour method to DBus
2020-04-28 11:10:32 | razer.device0.keymanager       | DEBUG    | Starting KeyWatcher
2020-04-28 11:10:32 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2020-04-28 11:10:32 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
[... cut ...]
2020-04-28 11:14:28 | razer.device0                  | DEBUG    | DBus call get_device_name
2020-04-28 11:14:28 | razer.device0                  | DEBUG    | DBus call get_device_name
2020-04-28 11:14:28 | razer.device0                  | DEBUG    | DBus call get_device_name
2020-04-28 11:14:28 | razer.device0                  | DEBUG    | DBus call get_brightness
2020-04-28 11:14:28 | razer.device0                  | DEBUG    | DBus call get_brightness
2020-04-28 11:14:28 | razer.device0                  | DEBUG    | DBus call get_firmware```

z3ntu, 2020-04-29:
> also reports faulty model name.

We're identifying devices by the USB PID which is `1532:0245` for yours I'm guessing? All devices with that PID (which might be both Advanced & Mercury of the 2019 Blade laptops) get identified with one name. If you can figure out how Synapse distinguishes between these products, we're happy to accept contributions.

> OpenRazer allows only Lightning style to be set

That's basically what we support in OpenRazer for any keyboard? What are you missing?

Abadaar, 2020-05-04:
I can confirm that the model, Razer Blade 2019 Advanced (black, 240Hz HD display), has `1532:0245` as USB PID.

z3ntu, 2020-05-07:
I can of course rename this model in the driver to "Razer Blade 2019 Advanced" but then the people with the Mercury version won't be happy.. CC @mgruppi 

Could you please run `sudo cat /sys/bus/hid/drivers/razerkbd/0*/test` which should output 3 bytes which might identify the model futher (the first byte contains the physical keyboard layout which we already use; I've also seen this call on mice before so it's probably some identification for hardware things?).

Abadaar, 2020-05-07:
This is my result: (I have a Nordic keyboard layout.)
070000

mgruppi, 2020-05-07:
> I can of course rename this model in the driver to "Razer Blade 2019 Advanced" but then the people with the Mercury version won't be happy.. CC @mgruppi
>
> Could you please run `sudo cat [...]

It is probably fine to rename the driver. PID 1532:0245 is the one for the Mid 2019 Advanced model just as 1532:023A is the one for the early 2019 Advanced Models (both Black and Mercury) but we report it as Advanced. 
Could we change the name and perhaps make it clear that it works for both Black and Mercury?

philippgerard, 2020-05-15:
Sorry for not getting back for so long and glad to see this got picked up by others, too.

Here's the result of your cat command:
```➜  ~ sudo cat /sys/bus/hid/drivers/razerkbd/0003:1532:0245.0004/test
03000```

The other two devices don't have a test command.

```➜  ~ ls -l /sys/bus/hid/drivers/razerkbd
total 0
lrwxrwxrwx 1 root root    0 15. Mai 12:08 0003:1532:0245.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0245.0002
lrwxrwxrwx 1 root root    0 15. Mai 12:08 0003:1532:0245.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0245.0003
lrwxrwxrwx 1 root root    0 15. Mai 12:08 0003:1532:0245.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0245.0004
--w------- 1 root root 4096 15. Mai 12:08 bind
lrwxrwxrwx 1 root root    0 15. Mai 12:08 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 15. Mai 12:08 new_id
--w------- 1 root root 4096 15. Mai 12:08 uevent
--w------- 1 root root 4096 15. Mai 12:08 unbind
```
Regarding features that I missed, I assumed it would be possible to get backlight support for the whole keyboard instead of just the function and arrow keys, same as in Windows.

z3ntu, 2020-05-19:
@mgruppi Could you please also run `sudo cat /sys/bus/hid/drivers/razerkbd/0*/test` on your laptop? I'm hoping the last two bytes can identify the device (the first byte is the keyboard layout as documented at https://github.com/openrazer/openrazer/wiki/Keyboard-layouts)

Abadaar, 2020-05-20:
I'm also submitting my out output from `ls -l /sys/bus/hid/drivers/razerkbd` in case it will help. I'm on the Razer Blade 2019 Advanced (black, 240Hz HD display) model, Nordic keyboard.
```
lrwxrwxrwx 1 root root    0 maj 20 07:01 0003:1532:0245.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0245.0007
lrwxrwxrwx 1 root root    0 maj 20 07:01 0003:1532:0245.0008 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0245.0008
lrwxrwxrwx 1 root root    0 maj 20 07:01 0003:1532:0245.0009 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0245.0009
--w------- 1 root root 4096 maj 20 07:51 bind
lrwxrwxrwx 1 root root    0 maj 20 07:01 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 maj 20 07:51 new_id
--w------- 1 root root 4096 maj 20 06:41 uevent
--w------- 1 root root 4096 maj 20 07:51 unbind
```

z3ntu, 2020-09-20:
Unfortunately I can't find out how to detect those different models from remote without having access to both machines. If you wish we can keep this issue open (indefinitely I guess?) until someone figures out something.

philippgerard, 2020-09-20:
Can we help from here? Provide some logs or dumps or something?

closed 2021-07-23 by github-actions[bot]