openrazer/openrazer #1122: Log is verbose even if it set to false

lah7, 2020-04-28:
Based on some discussion in #1106, I have noticed my `~/.local/share/openrazer/logs/razer.log` file grows into MBs over many months, but I have this set in `~/.config/openrazer/razer.conf`:

    verbose_logging = False

It might be a bug since there is a lot of `DEBUG` entries, which I wouldn't expect for a non-verbose log.

Logging could be improved, some ideas:

* Log recycling - old logs should be compressed after a certain file size (e.g. `razer.log.1.gz`)
* Verbose is on by default, this should be turned off, since generally these won't be read when the daemon is working normally.
* Custom effects are not being logged, should this be uncommented and print more useful details? (#1106)

z3ntu, 2020-04-28:
We pass `-v` to the daemon in https://github.com/openrazer/openrazer/blob/master/daemon/resources/openrazer-daemon.systemd.service.in#L8 so this could be a reason. Can you try removing it and see if that helps?

chabad360, 2020-04-29:
> Custom effects are not being logged, should this be uncommented and print more useful details? (#1106)

Perhaps we could have different levels of verbosity, this way that wouldn't be an issue. Question is if logger supports that.

Also, can we add back milliseconds to the logger?

lah7, 2020-04-30:
Thanks @z3ntu, yes - that was the reason. (162ba7b7493929da0219ef6de9e562478551ccca + bbb9187613fe1cb06dca860db8be92a1c2c365b0)

For log rotation, it turns out it's already enabled, at 16 MiB with 10 backup copies (theoretical max. 160 MiB!). I've reduced this to 1 MiB and 1 backup copy (theoretical max. 2 MiB) (8c69aaa162c4db3fc9525a168095beb24baa986b). Compression would be nice but requires more code and it's quite trivial at this point.

@chabad360 There already is `INFO`, `WARNING` and `DEBUG`, but with verbose always being "on", `DEBUG` was always being printed up to now.

Milliseconds could be useful, PRs welcome! (Or as part of your #1124 if it assists with those changes)

commit 162ba7b7 (2020-04-30): Remove verbose parameter for systemd service

Fixes #1122.

This has been overwriting the verbose_logging configuration option in
~/.config/openrazer/razer.conf

In most use cases, the daemon will be working normally so there's no
need for excess DEBUG logging in the system's journal or
~/.local/share/openrazer/logs/
commit bbb91876 (2020-04-30): daemon: Turn off verbose logging by default

Closes #1122.

- Verbose is for debugging, most users won't need this 'spam'
- Most users probably won't be reading the logs, so reduces the size of
on disk. For troubleshooting, it's advised to run -Fv anyway.
closed 2020-04-30 by lah7