openrazer/openrazer #1125: Missing Magic Sys Request implementation in kernel module razerkbd

Tikeri, 2020-04-30:
Hello,

The module is catching keyboard input events but does not handles the Magic Sys Request key combos which is pretty critical in some situations.

Running kernel version 4.4.164 and openrazer 2.7.0 (from git, merged in January).

Found that documentation but I am not sure how to implement this at all :
https://www.kernel.org/doc/html/latest/admin-guide/sysrq.html#i-want-to-add-sysrq-key-events-to-a-module-how-does-it-work

lah7, 2020-04-30:
From my understanding, the magic SysRq key always speaks directly to the kernel, so this isn't affected when `razerkbd` is loaded. I have it enabled on Manjaro and works as expected - e.g. flush cache to disk (Alt+SysRq+S)

The documentation you've found sounds useful if `razerkbd` wishes to add an input to Magic SysRq to handle a request, but I don't believe there is a use case for it right now. Good to know though!

Tikeri, 2020-04-30:
Thanks for the quick reply, well this is curious that it works on your system configuration.
I've done the following tests before opening this issue :
  1. razerkbd loaded : tried multiple SysRq and nothing happened + nothing from dmesg
  2. blacklisted razerkbd module + removed udev rule file from openrazer : after reboot I've tried the same and SysRq are working

I'm runnning Gentoo Linux and custom kernel but :
~ ;-) zgrep SYSR /proc/config.gz 
CONFIG_MAGIC_SYSRQ=y
CONFIG_MAGIC_SYSRQ_DEFAULT_ENABLE=0x1

lah7, 2020-04-30:
Whoops, I should have said it works for me with <kbd>Alt Gr</kbd> (right Alt) + <kbd>SysRq</kbd> + <kbd>S</kbd> but not with left <kbd>Alt</kbd>. I have a GB layout. :gb:

Now you mention it, I tried unloading the daemon and `razerkbd`, and using it with the left <kbd>Alt</kbd> combination works, so I guess that's the bug.

Does it work with your keyboard's right Alt key? (Easiest check is watching the kernel log and trying an emergency sync)

Tikeri, 2020-04-30:
I can confirm this behaviour, it works fine with Alt Gr here to (FR layout) :
[ 6768.484726] Emergency Sync complete

This means there is only a minor modification to do in the module source ? :)

Tikeri, 2020-04-30:
I am not sure to understand very from razerkbd_driver.c file :

lah7, 2020-05-02:
The latest release (2.8.0) has fixed the issue. Left ALT + SysRq works again. :+1:

closed 2020-05-02 by lah7