openrazer/openrazer #1126: Debounce interval handling

sandsmark, 2020-04-30:
Sorry that I didn't use the issue template, but I think this is a valid issue anyways.

I'm having trouble with registering the "correct" amount of clicks when running my mouse at 1000Hz report rate. I initially thought it was something mechanical, but adjusting the report rate in razergenie fixed it, and it only started happening after installing the openrazer module.

There are some [patches floating around](https://aur.archlinux.org/cgit/aur.git/tree/debounce.patch?h=xf86-input-evdev-debounce) as well as (some support)[https://aur.archlinux.org/packages/xf86-input-evdev-debounce/#comment-654099] in libinput, unfortunately with hardcoded values that don't seem appropriate for 1000Hz.

I also think it makes sense to handle debounce at the driver level.

I'm running the deathadder v2 branch that's not merged yet, but there's nothing touching the report rate or anything that could related to debouncing there.

So, am I missing something, or misunderstanding something? If people think it's a good idea to add some debounce handling in the driver, I can probably look into implementing it myself (unless someone else gets to it first).

z3ntu, 2020-05-01:
What do you mean by debouncing or debounce handling? We don't do anything with the actual input devices (apart on keyboards, blocking some keys while the game mode is active)

sandsmark, 2020-05-01:
Debounce is basically to filter out spurious events that happen because of too fast sampling (so a lowpass filter basically), so e. g. you get doubleclicks where there is only a single click. Or you "miss" release or press events.

A better explanation is here:
https://who-t.blogspot.com/2017/07/libinput-and-button-debouncing.html

Unfortunately libinput assumes a more normal sampling rate, at 1000Hz the events are not spaced with ~8ms (as described in that blog post).

So what I meant was that the openrazer driver could set up debounce when it bumps on the report rate. ButI (incorrectly) assumed that the debounce interval was configurable in the normal input drivers in the kernel, but it is just in e. g. gpio-keys, so this is a bit bigger task than what I thought.

But feel free to close this if it is out of scope (or my reasoning is wrong, I'm making some assumptions here).

z3ntu, 2020-05-02:
We don't actually handle the actual inputs for mice so I'd say this is out of scope.

closed 2020-05-02 by z3ntu