openrazer/openrazer #1132: Alt left registered as Control left

DasHammett, 2020-05-01:
### Distribution

Gentoo - Kernel 5.6.5

### Support

- [ ] Have you tried rebooting?
Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?
Yes

Hi, 

I have a BlackWidow Ultimate 2016 and the ALT left is not registered as such, but as Control left. Keyboard layout is Spanish (code 10) and is correctly identified by razerkbd.

I have used `xev` to capture the keystroke when pressing left ALT key to confirm what was being sent to the system.

```
KeyPress event, serial 70, synthetic NO, window 0x2400001,
    root 0x1e6, subw 0x0, time 1224348, (1078,5), root:(1085,981),
    state 0x10, keycode 37 (keysym 0xffe3, Control_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False
```
Could you perhaps review?
Thank you

Kind regards,

lah7, 2020-05-01:
It's a regression: https://github.com/openrazer/openrazer/commit/bd864c87c64908611cc989109c4970c14cf314fb - seems to only happen if the driver is loaded but the daemon is not running.

It might also be responsible for breaking SysRq with the left Alt key too (https://github.com/openrazer/openrazer/issues/1125)

Good job it's not a stable release yet, cos' this bug was winding me up thinking a KDE update had broken Alt+Tab. :sweat_smile:

DasHammett, 2020-05-01:
Glad to see it is already a known issue. 

Too bad in Gentoo systems there are no releases, it always pulls the repo.

Hope this can get resolved soon

Thanks

z3ntu, 2020-05-02:
Same issue as #1077 

I've reverted the patch on master for now, of course #872 isn't solved anymore now. I hope we find a good solution for that..

lah7, 2020-05-02:
I can confirm the latest release / master has fixed the Alt/Ctrl issue. (Or for users coming from a non-git 2.7 release, nothing broke at all :smile:) Please update your copy of OpenRazer - the version should read `2.8`.

closed 2020-05-02 by lah7

Tikeri, 2020-05-02:
Thanks a lot ! :)
I've upgraded to 2.8.0 already and got back the magic :)