openrazer/openrazer #1133: Support for Razer Huntsman Elite - matrix dimens: 9 22

svscorp, 2020-05-01:
### Device Info

* Name: Razer Huntsman Elite
* Product Number: RZ03-01870100-R3M1
* Model Number: RZ03-0187

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

===============
Bus 001 Device 006: ID 1532:0226 Razer USA, Ltd Razer Huntsman Elite
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0226 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
Please post the output of
    ls -lah /dev/input/by-id/

==============
drwxr-xr-x 2 root root 180  1 mei 21:28 .
drwxr-xr-x 4 root root 560  1 mei 21:28 ..
lrwxrwxrwx 1 root root  10  1 mei 21:00 usb-Logitech_USB_Optical_Mouse-event-mouse -> ../event17
lrwxrwxrwx 1 root root   9  1 mei 21:00 usb-Logitech_USB_Optical_Mouse-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10  1 mei 21:28 usb-Razer_Razer_Huntsman_Elite-event-if01 -> ../event20
lrwxrwxrwx 1 root root  10  1 mei 21:28 usb-Razer_Razer_Huntsman_Elite-event-kbd -> ../event18
lrwxrwxrwx 1 root root  10  1 mei 21:28 usb-Razer_Razer_Huntsman_Elite-if01-event-kbd -> ../event19
lrwxrwxrwx 1 root root  10  1 mei 21:28 usb-Razer_Razer_Huntsman_Elite-if02-event-mouse -> ../event20
lrwxrwxrwx 1 root root   9  1 mei 21:28 usb-Razer_Razer_Huntsman_Elite-if02-mouse -> ../mouse1
```

### Packet Captures

(let's see if we need it)

z3ntu, 2020-05-02:
I believe you got this message from RazerGenie, and your issue is covered in https://github.com/z3ntu/RazerGenie/issues/67

closed 2020-05-02 by z3ntu

svscorp, 2020-05-02:
Thanks, @z3ntu. Apologies for the wrong location of the issue created.