openrazer/openrazer #1134: Support for Razer Basilisk Essential

tomoki1337, 2020-05-02:
saw support added for the Razer Basilisk in #804fa58 yet still no support for the Essential model 
### Device Info

* Name: Razer Basilisk Essential 
* Product Number: RZ01-02650100
* Model Number: RZ01-0265

### Device Information

output of lsusb -d 1532:0065 -v

```
Bus 002 Device 004: ID 1532:0065 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0065 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Basilisk Essential
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```

```
lrwxrwxrwx 1 root root  10 May  2 22:20 usb-Razer_Razer_Basilisk_Essential-event-if01 -> ../event12
lrwxrwxrwx 1 root root   9 May  2 22:20 usb-Razer_Razer_Basilisk_Essential-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 May  2 22:20 usb-Razer_Razer_Basilisk_Essential-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 May  2 22:20 usb-Razer_Razer_Basilisk_Essential-if02-event-kbd -> ../event13
lrwxrwxrwx 1 root root   9 May  2 22:20 usb-Razer_Razer_Basilisk_Essential-mouse -> ../mouse0

```

### Packet Captures
currently not possible 
### Synapse Screenshots
currently not possible

tomoki1337, 2020-05-04:
added synapse screenshots

aabedraba, 2020-05-05:
Bumping this up, having the same problem.

rufex2001, 2020-06-08:
I would also like support for the Razer Basilisk Essential

simonyee, 2020-06-13:
Hi,
When is the development for the Razer Basilisk mouse for ubuntu 20.04 going to move forward?

Klaas-Lindi, 2020-07-29:
I would also like to have support for my essential! "surprisingly" the supported devices are the more expensive one...  What works for me is to install synapse on a windows device (might also work with OS.X), applying my settings and then connect the mouse to my linux-computer. I haven't tried a virtual windows computer or similar for this. Though the device still cannot be found by polychromatic afterwards, all the settings I have made are saved on the mouse, so it is fully functionally. Even the DPI button still has the values I programmed.

LinusPNutz, 2020-08-03:
I recently bought the mouse and would like to have it supported on Linux too.
The pulsing light kinda bugs me :) and i don't know how to change it on linux. It worked well with the keyboard.

Gimzie, 2020-09-05:
Just bought this same exact mouse as well, would be great to have support for it on Linux! +1

rcotrina94, 2020-09-17:
Just bought the Basilisk Essential. Let me know if I can help with the support for it in OpenRazer.

callihamm, 2020-11-30:
I see that at pcap has been requested, has this been provided yet? And if not, can you point where to provide this info?

z3ntu, 2020-11-30:
@callihamm https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

Roger-Roger-debug, 2020-12-08:
pcaps can be found on this openrgb [issue](https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/654)

z3ntu, 2020-12-20:
The pcaps contain info that `razer_chroma_extended_matrix_effect_custom_frame` is used with transaction id 0x1f, MATRIX_DIMS [1, 1], `razer_chroma_extended_matrix_brightness` with led id 0 (and of course also transaction id 0x1f).

If somebody wants to look into development for this device, you can probably start with a copy of the Mamba Elite and continue from there.

XECortex, 2021-01-16:
I've also got this mouse delivered today, and I thought it would work with the default Basilisk version driver, but then I saw this :(

How long will it take to add support for the Basilisk Essential mouse? I could also help reverse engineering if needed!

PR #1535 (2021-06-27): Add support for Razer Basilisk Essential
**Fixes #1134**

I've already tested everything on my PC, an additional review would be useful.

**Mouse settings:**
- DPI (Y-axis doesn't affect anything, but the option cannot be removed)
- Polling rate

**Logo effects:**
- Breath
- Spectrum
- Reactive
- Static
- None
closed 2021-06-27 by z3ntu