openrazer/openrazer #1139: Macros do not persist through login or reboot

dfarley1, 2020-05-03:
### Distribution

v2.8.0 through Ubuntu apt

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I configured the M1-5 keys on my BlackWidow Ultimate 2013 (firmware v3.0) to map to numpad 1-5. These work for the most part (there's definitely some lag/delay compared to normal keys), but the macros are not saved through a log out/in or reboot.

In the log there I can see the macros registering prior to reboot, but once the daemon comes back up it no longer prints anything when pressing the macro keys.
```
dan@dan-desktop:~$ tail -n 15 ~/.local/share/openrazer/logs/razer.log 
2020-05-03 11:04:10 | razer.device0.keymanager       | INFO     | Running Macro M1:[NP1 DOWN, NP1 UP]
2020-05-03 11:04:12 | razer.device0.keymanager       | INFO     | Running Macro M1:[NP1 DOWN, NP1 UP]
2020-05-03 11:04:13 | razer.device0.keymanager       | INFO     | Running Macro M1:[NP1 DOWN, NP1 UP]
2020-05-03 11:04:16 | razer.device0.keymanager       | INFO     | Running Macro M1:[NP1 DOWN, NP1 UP]
2020-05-03 11:04:17 | razer.device0.keymanager       | INFO     | Running Macro M1:[NP1 DOWN, NP1 UP]
2020-05-03 11:04:18 | razer.device0.keymanager       | INFO     | Running Macro M1:[NP1 DOWN, NP1 UP]
2020-05-03 11:04:18 | razer.device0.keymanager       | INFO     | Running Macro M1:[NP1 DOWN, NP1 UP]
2020-05-03 11:04:19 | razer.device0.keymanager       | INFO     | Running Macro M2:[NP2 DOWN, NP2 UP]
2020-05-03 11:04:20 | razer.device0.keymanager       | INFO     | Running Macro M2:[NP2 DOWN, NP2 UP]
2020-05-03 11:04:48 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 13623
2020-05-03 11:04:48 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-05-03 11:04:48 | razer                          | INFO     | Found device.0: 0003:1532:011A.0008
2020-05-03 11:04:48 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowUltimate2013
2020-05-03 11:04:48 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2020-05-03 11:04:48 | razer                          | INFO     | Serving DBus
```

I can re-save the macros but they disappear again after the next reboot.

dfarley1, 2020-05-03:
I just tried macros in a game and ran into a couple more issues.

1. Everything freezes for about 1/4 of a second when I press the macro button. This happens in-game and out; both Youtube and the game experience the freeze no matter which is in focus.
2. Assigning numpad keys to the macro doesn't work. I assigned M1 to Numpad1 but when I press M1 in-game it's detected as "normal" 1, not Numpad1. I tried with NumLock both enabled and disabled and neither worked. As in the original post the logs incorrectly state that "NP1" is being pressed. I confirmed this with an online keyboard tester as well.

z3ntu, 2020-05-04:
Macro support is pretty neglected in openrazer unfortunately. I have personally never really used macros so I've never touched the macro code.

chabad360, 2020-05-04:
In #1124 I'm working to add a (persistent) keybinding system that would support macros, however it does throw out the whole macro modifier (FN+F9) functionality (macros would only be configurable through DBus/GUI). Would that work for you, or would you like to be able to use the macro modifier as well?

dfarley1, 2020-05-04:
Personally I don't care about the FN+F9 functionality, that's just how Polychromatic told me to setup the macro keys. I just want the M1-5 keys to be mappable in games.

lah7, 2020-05-04:
@chabad360 Please do re-implement the _on-the-fly macro_ feature. I use them occasionally for repetitive editing tasks. It could be an alternate way to record macros into your WIP keybinding system and wouldn't regress existing functionality.

I guess I'm use to the current behaviour that they're "disposed" after each session, and I do like that for my use case. Though it could be improved further with timing delays, since the current macro feature doesn't always work in games (e.g. [T] to open chat + type command, or keys to open and type in a cheat code dialogue) since it's literally one keystroke after another - but it works well for bulk changes in a text file.

chabad360, 2020-05-04:
> Though it could be improved further with timing delays

@lah7 Overall implementing macros is relatively easy, however (recording) timing delays is another story. Probably the simplest way to do timing (for now), is to add them in manually. So I'll see what I can do.

closed 2020-10-04 by stale[bot]