openrazer/openrazer #1141: kali linux Support

joseluissaboya, 2020-05-05:
Please add Kali Linux support

honza-kasik, 2020-05-05:
@joseluissaboya Kali distribution is [Debian based](https://www.kali.org/docs/policy/kali-linux-relationship-with-debian/). Have you tried following [installation instructions](https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta) for Debian?

joseluissaboya, 2020-05-06:
I understand kali is debian based but when you go to the link it asks which operation system you are using. The options are Debian, Fendora, Ubuntu, Mageia, and openSUSE. When I select Debian it asks which Debian version: Stable, Testing, Debian 9.0 or 10.0
This is where I get stuck because I don't know if I choose the stable version if it would work since I am using an OS that is based on Debian but it's not the same.

honza-kasik, 2020-05-06:
> The Kali Linux distribution is based on Debian Testing.

— https://www.kali.org/docs/policy/kali-linux-relationship-with-debian/

JohnRGamesDesigner, 2020-05-25:
Hi there! I want to contribute to this thread - I have tried the the testing distro and it does not work unfortunately (see below):

root@razer:~# echo 'deb http://download.opensuse.org/repositories/hardware:/razer/Debian_Testing/ /' > /etc/apt/sources.list.d/hardware:razer.list
root@razer:~# wget -nv https://download.opensuse.org/repositories/hardware:razer/Debian_Testing/Release.key -O Release.key
2020-05-25 19:09:51 URL:https://download.opensuse.org/repositories/hardware:/razer/Debian_Testing/Release.key [1084/1084] -> "Release.key" [1]
root@razer:~# apt-key add - < Release.key
OK
root@razer:~# apt-get update
Get:1 http://download.opensuse.org/repositories/hardware:/razer/Debian_Testing  InRelease [1,531 B]
Get:2 http://download.opensuse.org/repositories/hardware:/razer/Debian_Testing  Packages [2,116 B]
Fetched 3,647 B in 0s (11.6 kB/s)    
Reading package lists... Done
root@razer:~# apt-get install openrazer-meta
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 openrazer-meta : Depends: openrazer-driver-dkms (= 2.8.0-0) but it is not going to be installed
                  Depends: openrazer-daemon (= 2.8.0-0) but it is not going to be installed
                  Depends: python3-openrazer (= 2.8.0-0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

JohnRGamesDesigner, 2020-05-25:
Further to the above, I just wanted to clarify I am using Kali linux as well:

root@razer:~# cat /etc/os-release
PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2020.2"
VERSION_ID="2020.2"
VERSION_CODENAME="kali-rolling"
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="https://www.kali.org/"
SUPPORT_URL="https://forums.kali.org/"
BUG_REPORT_URL="https://bugs.kali.org/"

JohnRGamesDesigner, 2020-05-25:
Ignore the above - my sources.list folder for some reason hadn't populated correctly during install - once I had linked the correct Kali mirrors and updated, it worked fine :)

z3ntu, 2020-06-12:
Can we close this now? The Debian packages should work fine on Kali.

Froazin, 2020-06-12:
Agreed :)

closed 2020-06-12 by z3ntu