openrazer/openrazer #1144: OpenRazer crashes X on Kubuntu 20.04

christianhujer, 2020-05-07:
On Kubuntu 20.04, when I login from SDDM, and OpenRazer is installed, X11 crashes, and I'm back on the SDDM login screen.

Attached Razer devices:
* Razer Blade Stealth 2017 (Core i7-8550U)
* Razer Core V2
* Razer Goliathus Extended
* Razer Firefly
* Razer Laptop stand Chroma
* Razer Headset stand Chroma
* Razer Man'O'War
* Razer Lancehead
* Razer Blackwidow Chroma

Probably irrelevant:
* Razer Stargazer
* Razer Kiyo
* Razer Seiren Pro
* Razer Leviathan

The user is in the `plugdev` group.

System modifications: None relevant, this is a fresh install of Kubuntu 20.04. There are a few additional packages installed like OpenJDK 13, GNU make, git, openssh-server, but nothing which I would even remotely see as a candidate for a conflict with OpenRazer.

I tried to see if there is any information in `~/.local/share/openrazer/logs/razer.log` but the file did not contain any useful information. I then tried setting `verbose_logging = True` in `~/.config/openrazer/razer.conf` but the directory remained empty after that change.

I can't debug this now, as I have to run an event on Saturday and need to get my system back up and running with Kubuntu 20.04 for that. But on Sunday, I might have time to debug this. So, any hints and instructions for debugging might be helpful.

z3ntu, 2020-05-07:
I believe that I've seen something similar recently. Having connected most of my Razer devices (BlackWidow Chroma, DeathAdder Chroma, Kraken Chroma, Firefly & Mamba Elite), and invoking Fn+F3 (Volume Up) shortcut would crash Xorg (and it also crashed sddm). After disconnecting one of my mice, it was working without any problems again. The Xorg log said something about in the input stack crashing iirc.

Does your crash also happen without you doing any input or does it crash when e.g. pressing keys on the keyboard?

christianhujer, 2020-05-07:
The crash that I describe happens without any input. I enter my password, hit enter, SDDM disappears, and SDDM appears again after a second, prompting me for my password again.

I also see stuff like Volume Up crashing X and even SDDM, but that's not what I'm talking about in this issue.

closed 2020-10-11 by stale[bot]