openrazer/openrazer #1147: Cynosa Chroma undetected by the driver

Mayravixx, 2020-05-12:
I'm at a bit of a loss here, I installed openrazer-meta from the AUR on my Arch installation today, and for whatever reason, Polychromatic claims that my supported razer keyboard isn't recognized. Anyone know how I could fix this? Is it also possible that it's doing this because I also have a HyperX mouse plugged in?
[image]

z3ntu, 2020-05-12:
Can you make sure your kernel headers are installed (e.g. `linux-headers`) and run 
`sudo dkms install openrazer-driver/2.8.0` ?

Mayravixx, 2020-05-12:
That was my issue, I didn't have the linux-headers package, and upon installing that, it ran sudo dkms install openrazer-driver/2.8.0 automatically. Closing this issue as it's now resolved, thanks.

closed 2020-05-12 by Mayravixx