openrazer/openrazer #1150: Support for Blade 15 Base Model (2020)

RedHatter, 2020-05-16:
Doesn't look like the 2020 blade is supported? I wouldn't think it should be too different from the 2019 model but what do I know. Will try and get pcaps tomorrow.

### Device Info

* Name: Blade 15 Base Model (2020)
* Product Number: RZ09-03287E72
* Model Number: RZ09-0328

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
```
Bus 001 Device 003: ID 1532:0255 Razer USA, Ltd Razer Blade
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0255
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)```

```
Please post the output of
    ls -lah /dev/input/by-id/
```
```
total 0
drwxr-xr-x 2 root root 160 May 15 21:19 .
drwxr-xr-x 4 root root 660 May 15 21:19 ..
lrwxrwxrwx 1 root root   9 May 15 21:19 usb-Azurewave_USB_Camera_0x0001-event-if00 -> ../event3
lrwxrwxrwx 1 root root   9 May 15 21:19 usb-Razer_Razer_Blade-event-if01 -> ../event9
lrwxrwxrwx 1 root root   9 May 15 21:19 usb-Razer_Razer_Blade-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 May 15 21:19 usb-Razer_Razer_Blade-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root  10 May 15 21:19 usb-Razer_Razer_Blade-if02-event-mouse -> ../event10
lrwxrwxrwx 1 root root   9 May 15 21:19 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

zuluhutu, 2020-06-02:
I got me a Razer Blade 15 Base Model 2020 last week. I decided to run ubuntu on it, after expierincing that debian did not support the hardware, yet. :( Network card and touchpad couldn't be found in the installer, and after installing it did not boot. just found your project. Would be very happy to use all the functions which it offers on ubuntu.

cmactavish, 2020-06-08:
I've also got a Razer Blade Base (2020), but the UK model. I would also appreciate support for this model!
### Device Info

   * Name: Blade 15 Base Model (2020)
   * Product Number: RZ09-03289W21
   * Model Number: RZ09-0328

### Device Information
(In case it's any different)

```
Please post the output of
    lsusb -d 1532: -v
```
```
Bus 001 Device 004: ID 1532:0255 Razer USA, Ltd Razer Blade
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0255 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```
```
Please post the output of
ls -lah /dev/input/by-id/
```
```
total 0
drwxr-xr-x 2 root root 220 Jun  8  2020 .
drwxr-xr-x 4 root root 720 Jun  8  2020 ..
lrwxrwxrwx 1 root root   9 Jun  8  2020 usb-Azurewave_USB_Camera_0x0001-event-if00 -> ../event5
lrwxrwxrwx 1 root root   9 Jun  8  2020 usb-Logitech_USB_Receiver-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Jun  8  2020 usb-Logitech_USB_Receiver-if01-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Jun  8  2020 usb-Logitech_USB_Receiver-if01-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Jun  8  2020 usb-Razer_Razer_Blade-event-if01 -> ../event13
lrwxrwxrwx 1 root root  10 Jun  8  2020 usb-Razer_Razer_Blade-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 Jun  8  2020 usb-Razer_Razer_Blade-if01-event-kbd -> ../event12
lrwxrwxrwx 1 root root  10 Jun  8  2020 usb-Razer_Razer_Blade-if02-event-mouse -> ../event16
lrwxrwxrwx 1 root root   9 Jun  8  2020 usb-Razer_Razer_Blade-if02-mouse -> ../mouse3
```

z3ntu, 2020-06-12:
If any of you is comfortable with editing some files, you can follow the changes done for other Blade laptops (see recent PRs or commits) for this device and figure out what works.

z3ntu, 2020-07-18:
A PR is available at #1171 , would be nice if you could test that.

plucked, 2020-08-13:
Hey, I can check it out, but you have to help me in how to install? I tried using the make file, but it breaks my python setup. (Ubuntu 20.04 on a fresh system, so it did not hurt)

vvvrrooomm, 2020-08-13:
It needs a few python packages installed, maybe those are missing? The required packages are:
   "numpy" "python"  "PyGObject" "setproctitle" "pyudev"
you might want to use a [venv](https://docs.python.org/3/library/venv.html) for those packages

zhassanbey, 2020-08-14:
Hi!
I have cloned the branch from the fork.
Then I tried installing it using `sudo make daemon_install`.
 The driver has been installed but now I am seeing something like `/blah_blah/device_type, file is not owned by plugdev` log.

I am pasting below the full log from `openrazer-daemon -Fv`:
`
Starting daemon.
2020-08-13 22:57:33 | razer                          | INFO     | Initialising Daemon (v2.8.0). Pid: 3770
2020-08-13 22:57:33 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Base ---------------------- (1532:023B)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Mercury ------------------- (1532:0240)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Adv ----------------------- (1532:023A)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Base ---------------------- (1532:0246)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019StudioEdition ------------- (1532:024D)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeEarly2020Base ----------------- (1532:0255)
2020-08-13 22:57:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
[... cut ...]
2020-08-13 22:57:33 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2020-08-13 22:57:33 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2020-08-13 22:57:33 | razer                          | INFO     | Serving DBus
^C2020-08-13 22:57:43 | razer                          | INFO     | Stopping daemon on signal 2
Stopping daemon.
` 

I also have tried to remove all the previously installed daemon and reinstalling it, seeing the same error.

Please help.

vvvrrooomm, 2020-08-16:
Saw it as well. The problem is with the razer_mount script and udev rule which only acts on HID devices - somehow this device is not classified as HID. A fix is now on this feature branch, not sure if this is a proper way to fix it

zhassanbey, 2020-08-17:
> Saw it as well. The problem is with the razer_mount script and udev rule which only acts on HID devices - somehow this device is not classified as HID. A fix is now on this feature branch, not sure if this is a proper way to fix it

Thanks for the reply.
When I just pulled from the branch and run ```openrazer-daemon -Fv``` the problem the device was recognized,
but later after I reboot the problem came back. I tried again uninstalling and installing again, but problem remained.

Any ideas why?
Also how is this gonna work with frontend app like polychromatic or razercommander? Should we just remove openrazer installation line in the PKGBUILD?

vvvrrooomm, 2020-08-17:
I think the updated udev rules file must be integrated into the initrd image. On debian I did `update-initramfs -k all -c` to recreate these. 
Polychromatic should work out of the box - a few effects appear which are not supported - apart from that it shoud be good. No clue about razercommander, I don't use it

zhassanbey, 2020-08-17:
Thanks for the help!
Today I run ```sudo make install_i_know_what_i_am_doing``` and the the daemon service just worked fine.

P.S: BTW it would be nice to add some packaging script to package daemon and python libs too, cause polychromatic in turn did not just work out of the box from the package manager. Therefore I went to the polychromatic repository and built it with make from source as well. Now everything works!

FranciscodeMaussion, 2020-09-24:
Have tried this for some days, everything works great. Maybe the effects that doesnt apply should be hidden.

PR #1171 (2020-09-26): Add support for Razer Blade 15 Early 2020 base
Fixes #1150
closed 2020-09-26 by z3ntu