openrazer/openrazer #1151: Support for Razer Blade 2018 - en_GB keyboard layout

danryu, 2020-05-18:
Hi, neither Polychromatic or RazerGenie allow me to fully configure the keyboard on my Razer Blade 2018 - with UK (GB) keyboard layout.

RazerGenie doesn't Some keys are unable to be mapped in Polychromatic - 
- both Alt keys
- up cursor key
- right Shift key
- Enter key
- F1 key

Also I would like to be able to toggle the "fn" Function key to allow eg single-press volume key operation.

Please let me know if is an openrazer issue, or to be separately dealt with in RazerGenie/Polychromatic.

Thanks!
---

### Device Info

* Name: Razer Blade 2018
* Product Number: RZ09-02385W71
* Model Number: RZ09-02385

### Device Information

Bus 001 Device 003: ID 1532:0233 Razer USA, Ltd Razer Blade
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0233 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```

lrwxrwxrwx 1 root root   9 May 18 19:57 usb-Razer_Razer_Blade-event-kbd -> ../event7
lrwxrwxrwx 1 root root  10 May 18 19:57 usb-Razer_Razer_Blade-if01-event-kbd -> ../event14
lrwxrwxrwx 1 root root   9 May 18 19:57 usb-Razer_Razer_Blade-if02-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 May 18 19:57 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

chabad360, 2020-05-18:
Can you run `evtest` on your keyboard, hit the <kbd>2</kbd> key, and paste the output?

Output should look similar to this:

```
Input driver version is 1.0.1
Input device ID: bus 0x11 vendor 0x1 product 0x1 version 0xab41
Input device name: "AT Translated Set 2 keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    ...
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
Testing ... (interrupt to exit)
Event: time 1589833362.472996, type 4 (EV_MSC), code 4 (MSC_SCAN), value 03
Event: time 1589833362.472996, type 1 (EV_KEY), code 3 (KEY_2), value 1
Event: time 1589833362.472996, -------------- SYN_REPORT ------------
2Event: time 1589833362.564741, type 4 (EV_MSC), code 4 (MSC_SCAN), value 03
Event: time 1589833362.564741, type 1 (EV_KEY), code 3 (KEY_2), value 0
Event: time 1589833362.564741, -------------- SYN_REPORT ------------
```

danryu, 2020-05-18:
Sure...

```$ sudo evtest

Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x1 version 0x0
Input device name: "Power Button"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 116 (KEY_POWER)
Properties:
Testing ... (interrupt to exit)
```

No events show up ....

danryu, 2020-05-18:
`evdev` shows identical devices, which is confusing:

```Available devices:
/dev/input/event7:      Razer Razer Blade
/dev/input/event9:      Razer Razer Blade
/dev/input/event14:     Razer Razer Blade
```

Trial and error suggests that 14 is the keyboard. Below is the output: 

```
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x1532 product 0x233 version 0x111
Input device name: "Razer Razer Blade"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 34 (KEY_G)
    Event code 35 (KEY_H)
    Event code 36 (KEY_J)
    Event code 37 (KEY_K)
    Event code 38 (KEY_L)
    Event code 39 (KEY_SEMICOLON)
    Event code 40 (KEY_APOSTROPHE)
    Event code 41 (KEY_GRAVE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 44 (KEY_Z)
    Event code 45 (KEY_X)
    Event code 46 (KEY_C)
    Event code 47 (KEY_V)
    Event code 48 (KEY_B)
    Event code 49 (KEY_N)
    Event code 50 (KEY_M)
    Event code 51 (KEY_COMMA)
    Event code 52 (KEY_DOT)
    Event code 53 (KEY_SLASH)
[... 214 lines cut ...]
      Max      572
    Event code 40 (ABS_MISC)
      Value      0
      Min        0
      Max      255
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    250
    Repeat code 1 (REP_PERIOD)
      Value     33
Properties:
Testing ... (interrupt to exit)
Event: time 1589836386.215793, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70028
Event: time 1589836386.215793, type 1 (EV_KEY), code 28 (KEY_ENTER), value 0
Event: time 1589836386.215793, -------------- SYN_REPORT ------------
Event: time 1589836388.164762, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7001f
Event: time 1589836388.164762, type 1 (EV_KEY), code 3 (KEY_2), value 1
Event: time 1589836388.164762, -------------- SYN_REPORT ------------
2Event: time 1589836388.274755, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7001f
Event: time 1589836388.274755, type 1 (EV_KEY), code 3 (KEY_2), value 0
Event: time 1589836388.274755, -------------- SYN_REPORT ------------
```

danryu, 2020-05-18:
> > RazerGenie doesn't Some keys are unable to be mapped in Polychromatic -
> 
> Do they just not allow it, or is there an error that's given as well?

RazerGenie doesn't allow it, just shows a dialog with the text:
_"Your keyboard layout (en_GB) is not yet supported by RazerGenie for this keyboard. Please open an issue in the RazerGenie repository."_

(Yes, I will open an issue with them as soon as I establish there isn't an upstream error!)

Polychromatic shows a fully-mapped keyboard in the profile editor, but still the aforementioned keys are not physically illuminated.

chabad360, 2020-05-18:
> `evtest` shows 3 identical-looking devices, which is confusing:

One for power button, mouse, and the keyboard.

What distro are you using?

danryu, 2020-05-18:
> What distro are you using?

Kubuntu 20.04

chabad360, 2020-05-18:
Run the following script:

```python
from openrazer.client import DeviceManager()

a = DeviceManager()
b = a.devices[0]

b.fx.advanced.matrix[0, 3] = (255, 0, 0)
b.fx.advanced.matrix[5, 3] = (255, 0, 0)
b.fx.advanced.matrix[5, 11] = (255, 0, 0)
b.fx.advanced.matrix[4, 14] = (255, 0, 0)
b.fx.advanced.matrix[4, 16] = (255, 0, 0)
b.fx.advanced.matrix[3, 14] = (255, 0, 0)
b.fx.advanced.draw()
```

That script should make all the keys you listed as broken Red, if that doesn't happen its either an Openrazer problem or your hardware is broken (unlikely), otherwise its a problem with polychromatic.

danryu, 2020-05-18:
> ```python
> from openrazer.client import DeviceManager()
> ```

After removing () braces in line 1 it now runs like so:

```
$ python3 test.py 
Traceback (most recent call last):
  File "test.py", line 10, in <module>
    b.fx.advanced.matrix[4, 16] = (255, 0, 0)
  File "/usr/lib/python3/dist-packages/openrazer/client/fx.py", line 989, in __setitem__
    assert 0 <= key[1] < self._cols, "Column out of bounds"
AssertionError: Column out of bounds
```

chabad360, 2020-05-18:
Just take out the offending line for now.

danryu, 2020-05-18:
> Just take out the offending line for now. Although that shouldn't happen...

Ok that runs now, but it just turned the following 3 keys red:
- Windows key
- left Ctrl key
- F2 key

:D

chabad360, 2020-05-18:
> Ok that runs now, but it just turned the following 3 keys red:
> 
>     * Windows key
> 
>     * right Ctrl key
> 
>     * F2 key

Weird, lets try a few different keys:

```python
b.fx.advanced.matrix[0, 2] = (255, 0, 0)
b.fx.advanced.matrix[5, 2] = (255, 0, 0)
b.fx.advanced.matrix[5, 10] = (255, 0, 0)
b.fx.advanced.matrix[4, 13] = (255, 0, 0)
b.fx.advanced.matrix[4, 15] = (255, 0, 0)
b.fx.advanced.matrix[3, 13] = (255, 0, 0)
```

danryu, 2020-05-18:
> Weird, lets try a few different keys:

This feels like a game of Battleship ..! lol

Anyway you got 3 hits with that last one!
- F1
- up cursor key
- right Shift key

Also lit the Fn key and #, but they were already working.
So both Alt keys and Enter are still outstanding.

But I guess this means that the problems are in Polychromatic/RazerGenie, right?
So I will log issues accordingly.

Lastly - is the Fn toggle feature (to negate need to press Fn key for eg Volume functions) something to be supported in openrazer or in Polychromatic/RazerGenie?
Thanks a lot for your help btw

chabad360, 2020-05-18:
> Lastly - is the Fn toggle feature (to negate need to press Fn key for eg Volume functions) something to be supported in openrazer or in Polychromatic/RazerGenie?

Well, it would be in the domain of Openrazer. Thing is, on most keyboards the <kbd>FN</kbd> key is a hardware key, and there's no way to tell from software if it's pressed. To my knowledge, there are some Razer keyboards that do this, but I'm not sure which. But, my PR #1124 would allow for you to just map the respective Fn key to the volume keys.

> Thanks a lot for your help btw

You're welcome 😊

danryu, 2020-05-19:
Regarding the Fn toggle issue - I'm pretty sure I have seen this previously in OpenRazer/Polychromatic.

There is relevant code in openrazer/razerkbd_drv.c:
https://github.com/openrazer/openrazer/blob/72fc42cc334d79efdc83249fd3af69e9e676dbf2/driver/razerkbd_driver.c#L1547

chabad360, 2020-05-19:
Have a look in your device files (`/sys/bus/hid/drivers/razerkbd/<DEVICE ID>/`) if there is a file called `fn_toggle`.

danryu, 2020-05-19:
> Have a look in your device files (`/sys/bus/hid/drivers/razerkbd/<DEVICE ID>/`) if there is a file called `fn_toggle`.

Unfortunately not. Do you know why this is?

```
/sys/bus/hid/drivers/razerkbd$ find */ | grep fn_toggle
/sys/bus/hid/drivers/razerkbd$
```

chabad360, 2020-05-19:
Likely, the driver doesn't provide it for your device. You can try to add
it manually, and see what happens (I can give some pointers tomorrow, I'm
too tired to look now).

danryu, 2020-05-19:
Thanks, that would be great. I'm happy to hack a bit and create a PR if you can give me those pointers....

Sleep well, mate ;)

chabad360, 2020-05-20:
@danryu https://github.com/openrazer/openrazer/blob/72fc42cc334d79efdc83249fd3af69e9e676dbf2/driver/razerkbd_driver.c#L2396-L2413

and https://github.com/openrazer/openrazer/blob/72fc42cc334d79efdc83249fd3af69e9e676dbf2/driver/razerkbd_driver.c#L2392

It's possible you'll need to add another `case`, and it's also possible that it's not supported for your device (I find it odd that such a feature would be included on older devices and not newer ones). 

Have Fun!

danryu, 2020-05-20:
@chabad360 

If by "supported on my device" you mean that there is a hardware requirement to make the Fn toggle feature work - that _must_ be supported on my device as the feature works fine under windows, configurable under Razer Synapse.

As far as I can tell, it's a single line change to implement the fn_toggle - shown here in my fork:

https://github.com/danryu/openrazer/blob/cd516f0067e8f6ab4599239b281d88e21cf2d928/driver/razerkbd_driver.c#L2412

https://github.com/danryu/openrazer/commit/cd516f0067e8f6ab4599239b281d88e21cf2d928

I'm not sure if adding another case here would be useful. I haven't for now.

I'm following the build instructions for Ubuntu as at https://github.com/openrazer/openrazer/wiki/Building-a-package
but the build script is failing as pasted below. Would be grateful if you could provide any more info on build deps etc, to reduce my debugging time. I already have an [open issue](https://gitlab.freedesktop.org/libinput/libinput/-/issues/489) with wayland.freedesktop.org to get libinput to allow palm rejection on the touchpad! :)

Final output of `./scripts/build_debs.sh`:

```:: Installing for Ubuntu
=====================i================================
mv /tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/tmp/usr/lib/python3.* /tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/tmp/usr/lib/python3
mv /tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/tmp/usr/lib/python3/site-packages /tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/tmp/usr/lib/python3/dist-packages
install: creating directory '/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/tmp/usr/share/dbus-1'
install: creating directory '/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/tmp/usr/share/dbus-1/services'
'org.razer.service' -> '/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/tmp/usr/share/dbus-1/services/org.razer.service'
install: creating directory '/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/tmp/usr/lib/systemd'
install: creating directory '/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/tmp/usr/lib/systemd/user'
'openrazer-daemon.service' -> '/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/tmp/usr/lib/systemd/user/openrazer-daemon.service'
make[2]: Leaving directory '/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer'
make[1]: Leaving directory '/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer'
   debian/rules override_dh_install
make[1]: Entering directory '/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer'
dh_install --package=openrazer-daemon usr/share/dbus-1/services/org.razer.service
/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/openrazer-driver-dkms.install: 1: lib/udev/: not found
/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/openrazer-driver-dkms.install: 2: usr/src/: not found
/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer/debian/openrazer-driver-dkms.install: 3: usr/share/metainfo/io.github.openrazer.openrazer.metainfo.xml: not found
dh_install: warning: debian/openrazer-driver-dkms.install is marked executable but does not appear to an executable config.
dh_install: warning: 
dh_install: warning: If debian/openrazer-driver-dkms.install is intended to be an executable config file, please ensure it can
dh_install: warning: be run as a stand-alone script/program (e.g. "./debian/openrazer-driver-dkms.install")
dh_install: warning: Otherwise, please remove the executable bit from the file (e.g. chmod -x "debian/openrazer-driver-dkms.install")
dh_install: warning: 
dh_install: warning: Please see "Executable debhelper config files" in debhelper(7) for more information.
dh_install: warning: 
dh_install: error: debian/openrazer-driver-dkms.install (executable config) returned exit code 127
make[1]: *** [debian/rules:34: override_dh_install] Error 25
make[1]: Leaving directory '/tmp/tmp.q2Gkx9tV5c_deb_build_tmp/razer'
make: *** [debian/rules:13: binary] Error 2
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
```

chabad360, 2020-05-20:
From what it seems, you just need to run `chmod -x "debian/openrazer-driver-dkms.install` and commit. As I don't think that's meant to be ran. But I can't confirm as I use Arch...

danryu, 2020-05-20:
Sigh - was just a permissions problem caused by a bad mount. (facepalm)

Ok the drivers are built now. What's the advice on how to test-install them?
Re versioning etc. Should the main drivers be purged from the system first?

danryu, 2020-05-20:
> Re versioning etc. Should the main drivers be purged from the system first?

So I did a purge and my newly built drivers are running. Polychromatic is running as well.

However, I had assumed Polychromatic would reveal the fn_toggle feature when running with a driver/device that supported it - but it isn't the case.
How can I test whether the fn_toggle actually works?

chabad360, 2020-05-20:
> How can I test whether the fn_toggle actually works?

Should go something like:

```sh
echo 1 |  sudo tee /sys/bus/hid/drivers/razerkbd/<DEVICE ID>/fn_toggle
```

danryu, 2020-05-21:
> ```shell
> echo 1 |  sudo tee /sys/bus/hid/drivers/razerkbd/<DEVICE ID>/fn_toggle
> ```

Yep, so that doesn't work.

> It doesn't seem to be implemented in the daemon.

Not sure I get this. Why are the fn_toggles in there then? Is it half-implemented?
And if do, how much work would it be to get it in?

I can't believe that I'm the only Linux-on-Blade2018 user out there who cares about this...

I feel unlucky. On my Blade 2018 with Linux:
- reboot stalls, suspend doesn't resume (tried a number of grub hacks, to no avail)
- touchpad palm rejection doesn't work, making the keyboard _incredibly_ difficult to use
- fn_toggle doesn't work, making all Volume changes and Brightness changes an annoying 2-handed affair

All these usability hindrances make the whole experience a drag when it could be beautiful :((

danryu, 2020-05-23:
Closing this as I digressed from original issue title - which is not a problem in openrazer

closed 2020-05-23 by danryu

MatzE-sch, 2020-10-12:
For some other googlers landing here: I have the German keyboard layout with the razer blade 15

also my "arrow up", "enter" and "left shift" did not light up with polychromatic and razerGenie.

I created a mapping for all keys thanks to the help of 

> Run the following script:
>
> ```python
> from openrazer.client import DeviceManager
>
> a = DeviceManager()
> b = a.devices[0]
>
> b.fx.advanced.matrix[0, 3] = (255, 0, 0)
> b.fx.advanced.matrix[5, 3] = (255, 0, [...]

```python3 
from openrazer.client import DeviceManager
mapping = {
    "esc": (1, 0),
    "f1": (2, 0),
    "f2": (3, 0),
    "f3": (4, 0),
    "f4": (5, 0),
    "f5": (6, 0),
    "f6": (7, 0),
    "f7": (8, 0),
    "f8": (9, 0),
    "f9": (10, 0),
    "f10": (11, 0),
    "f11": (12, 0),
    "12": (13, 0),
    "ins": (14, 0),
    "del": (15, 0),
    "backspace": (15, 1),
    "r shift": (15, 4),
    "r fn": (15, 5),
    "arrow left": (14, 5),
    "arrow down": (13, 5),
    "arrow left": (12, 5),
    "r strg": (11, 5),
    "alt gr": (9, 5),
    "alt": (5, 5),
    "super": (3, 5),
    "l fn": (2, 5),
    "l ctrl": (1, 5),
    "r shift": (1, 4),
    "caps lock": (1, 3),
    "tab": (1, 2),
    "^": (1, 1),
    "1": (2, 1),
    "2": (3, 1),
    "3": (4, 1),
    "4": (5, 1),
    "5": (6, 1),
    "6": (7, 1),
    "7": (8, 1),
    "8": (9, 1),
    "9": (10, 1),
    "0": (11, 1),
    "ß": (12, 1),
    "´": (13, 1),
    "+": (13, 2),
    "#": (13, 3),
    "arrow up": (13, 4),
    "enter": (14, 2),
    "ü": (12, 2),
    "p": (11, 2),
    "o": (10, 2),
    "i": (9, 2),
    "u": (8, 2),
    "z": (7, 2),
    "t": (6, 2),
    "r": (5, 2),
    "e": (4, 2),
    "w": (3, 2),
    "q": (2, 2),
    "a": (2, 3),
    "<": (2, 4),
    "y": (3, 4),
    "x": (4, 4),
    "c": (5, 4),
    "v": (6, 4),
    "b": (7, 4),
    "n": (8, 4),
    "m": (9, 4),
    ",": (10, 4),
    ".": (11, 4),
    "-": (12, 4),
    "ä": (12, 3),
    "ö": (11, 3),
    "l": (10, 3),
    "k": (9, 3),
    "j": (8, 3),
    "h": (7, 3),
    "g": (6, 3),
    "f": (5, 3),
    "d": (4, 3),
    "s": (3, 3)
}
coords = mapping['arrow up']
b.fx.advanced.matrix[coords[1],coords[0]] = (255, 0, 0)
```