openrazer/openrazer #1153: Driver sometimes loads incorrect on boot

leofah, 2020-05-21:
### Distribution

Arch Linux x86_64
Kernel: linux 5.6.13.arch1-1

### Support

- [ ] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Sometimes after boot the driver is not loaded correctly (I guess). If i want to press a Key, this key will be pressed multiple times (like 15 times). It look likes the key is released to late.This makes it impossible to log in or reboot with the keyboard. However after re-plugging the keyboard it works fine with no problems.

I think this happened after each reboot if a new kernel was installed and the dkms had run. But on the last time this happened the last kernel update was a while ago.

Here is my dmesg and journalctl log, if it helps.
[razer.log](https://github.com/openrazer/openrazer/files/4661323/razer.log)

z3ntu, 2020-06-12:
Does this issue also happen with openrazer *not* installed?

leofah, 2020-06-22:
After uninstalling the driver this issue happened directly on the next boot, but since then, it has not happened the past few days.

chabad360, 2020-06-22:
Is it possible that a driver issue can include the brightness being set to zero?

closed 2021-06-21 by github-actions[bot]