openrazer/openrazer #1154: Request: fn_toggle support on Razer Blade 15 2018

danryu, 2020-05-23:
### Distribution

Kubuntu 20.04 64bit

### Support

Please enable **fn_toggle** feature on a Razer Blade 15 2018.

I have made a single-line change here
https://github.com/danryu/openrazer/commit/cd516f0067e8f6ab4599239b281d88e21cf2d928
to create fn_toggle file under /sys, but the following command results in no change:

`echo -n "1" > /sys/bus/hid/drivers/razerkbd/0003\:1532\:0233.0004/fn_toggle`

Let me know what debug I can provide to help make this happen, or please point me in the right direction to start generating a PR....
Thanks!

references: https://github.com/openrazer/openrazer/issues/1151
also: https://github.com/openrazer/openrazer/issues/545

danryu, 2020-06-11:
I appreciate that Fn is in a different category of key, but is this really considered completely unimportant by all Razer users on Linux? (I'm assuming that openrazer doesn't support fn_toggle on any Razer model, from what you say.)

Does every Linux-on-Razer user here really enjoy using 2 fingers on 2 hands to change the volume, change the brightness, change the keyboard backlight brightness? I'm really surprised there isn't more interest in this here....

Let me guess - you don't need to change the volume or brightness, because you've made your setup work round that limitation?

chabad360, 2020-06-11:
Honestly, it never bothered me that much, and I don't own a Razer laptop or keyboard (just a keypad). But I agree that it's definitely a feature that should be supported. Unfortunately, my knowledge of C is nil, so I can't exactly do anything about it...

danryu, 2020-06-11:
I don't mind hacking a bit of C to get something like this working, but diving into the codebase without prior expoure isn't something I can do now with my schedule.

From the commit history, it looks like @z3ntu is maintaining much of the relevant driver code.
@z3ntu if you have a minute - what is your take on implementing fn_toggle support?

If you can give me an idea of the scale of the required modification - and if you want to give some quick advice on implementation - that would be great.

z3ntu, 2020-06-12:
It looks like we're just sending the value given to the fn_toggle sysfs file to the device:
https://github.com/openrazer/openrazer/blob/72fc42cc334d79efdc83249fd3af69e9e676dbf2/driver/razerkbd_driver.c#L1547-L1556

I'm actually not sure myself what the toggle does - never used it I think

danryu, 2020-06-13:
Hi @z3ntu, thanks for dropping in!

The following assumes we are referring to a Razer Blade laptop (do you have one of these by any chance?)
If I'm not mistaken, the "fn_toggle" refers to the Function key being "toggled" to its inverse state, meaning that the Fn key does *not* have to be pressed to operate the multimedia section, ie volume up/down can be pressed with F2 and F3 only, screen brightness up/down with F8/F9 only, and so on. 

In the Windows app Synapse, the feature is referred to as "Function Key Primary".
It would be a MASSIVE help to me (and many other Linux Blade users I'm sure) to have this enabled, as it makes using those essential functions in different environments MUCH easier - operating the Mute button with one finger, for example, would be very handy!

I've attached a pic of the Synapse feature so you can see the context.

[image]

Please let me know what you think the chances are of getting this done!
Thanks mate

z3ntu, 2020-06-26:
Interesting, thanks for the context! And no I don't own a Razer laptop :) There are actually quite a few features on the hundreds of Razer devices where I'm not sure what they're actually used for :D

Please for RazerGenie open a new issue in https://github.com/z3ntu/RazerGenie/issues with your screenshot, and I'm guessing https://github.com/polychromatic/polychromatic/issues/218 is also about this? If it's not then please also open an issue there so it can get implemented :slightly_smiling_face:

lah7, 2021-03-04:
Just to confirm, has this been implemented in the driver (as a `sysfs` file) and does it work?

A [quick repository search suggests it's there](https://github.com/openrazer/openrazer/search?q=toggle), but not exposed in the API. If it is an `echo 1 > /sys/.../set_fn_toggle` then what needs to be echo'd to switch between the two?

danryu, 2021-03-13:
@lah7 
I just had my Blade 15 2018 replaced for an early 2020 Blade 15, and the problem is the same - there's no sysfs file.

Bus 001 Device 003: ID 1532:0255 Razer USA, Ltd Razer Blade
Using openrazer 2.9.0~ubuntu20.04.1  on Ubuntu 20.04/

Would love any kind of hack or workaround ....

z3ntu, 2021-03-14:
@danryu You can try modifying the driver and add `CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_fn_toggle);` to https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L2379 and `device_remove_file(&hdev->dev, &dev_attr_fn_toggle);` to https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L2701 , then rebuild the driver and load it. Then the sysfs file should appear and you can try writing to it and see if something happens.

danryu, 2021-03-15:
@z3ntu 
I made the hacks as suggested...
https://github.com/danryu/openrazer/commit/b9ae83340499f8e09945072057636b5e7040a181

Rebuilt - I used `scripts/build_debs.sh`, purged all existing openrazer debs from installation, rebooted, installed the newly built debs from /dist/, rebooted.

The sysfs fn_toggle file is now in place.
However
`echo 1 >  /sys/bus/hid/drivers/razerkbd/0003\:1532\:0255.0004/fn_toggle`
apparently has no effect. I still need the Fn key pressed for all media functions.

z3ntu, 2021-03-21:
Too bad, was worth a shot.

Another try would be, that you record usb pcaps of Razer Synapse changing the setting multiple times (before starting the capture, set the lighting effects to disabled) and I can take a look at them. But it's possible that Synapse does this all in software instead of in firmware, but we'll know with pcaps.

danryu, 2021-07-21:
Bumping this for now - still an open issue as far as I am concerned.

Unfortunately I haven't been able to do do the pcap on Windows yet as @z3ntu suggested - the Razer Blade is out of action after multiple power supplies failed. 
I will update when I am back online with a new PSU.

danryu, 2021-09-22:
I can't even test the above, as no Razer hardware I own ends up lasting more than a couple of weeks. The power bricks in particular for the Razer Blades. So I am returning the Blade while I still can, and never, ever buying a Razer again. 
So I guess I'll close this issue out :)

closed 2021-09-22 by danryu