openrazer/openrazer #1156: Support for Razer Huntsman Elite keyboard

mgrendahl, 2020-05-27:
### Device Info

* Name: Razer Huntsman Elite Keyboard
* Product Number: R203-01870200-R3U1
* Model Number: R203-0187

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

Bus 009 Device 005: ID 1532:0226 Razer USA, Ltd Razer Huntsman Elite
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0226 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Huntsman Elite
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

```
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

I installed openrazer:
python3 -m pip install openrazer

root@lucy:/home/martin/razer# /usr/bin/razercli
Usage: razercli [OPTIONS] COMMAND [ARGS]...

  Razerkeyboard CLI

Options:
  --help  Show this message and exit.

Commands:
  off     Set the mode to static, default color ff00ff
  static  Set the mode to static, default color ff00ff
root@lucy:/home/martin/razer# /usr/bin/razercli static
Traceback (most recent call last):
  File "/usr/bin/razercli", line 8, in <module>
    sys.exit(razer_cli())
  File "/usr/lib64/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib64/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/lib64/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib64/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib64/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib64/python3.8/site-packages/openrazer/cli.py", line 16, in static
    keyboard = Keyboard()
  File "/usr/lib64/python3.8/site-packages/openrazer/keyboard.py", line 41, in __init__
    self.device = found_devices[0]
IndexError: list index out of range
root@lucy:/home/martin/razer# 

I'd like to set it static green.
It would be nice to be able to use the driver interface too.  It is supported for various operating systems like debian, suse, etc, but not for slackware64 (current) which I use ...

mgrendahl, 2020-05-27:
ls -lah /dev/input/by-id/
root@lucy:/home/martin/razer# ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 180 May 27 13:43 ./
drwxr-xr-x 4 root root 440 May 27 13:43 ../
lrwxrwxrwx 1 root root   9 May 12 19:15 usb-Logitech_USB_Laser_Mouse-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 May 12 19:15 usb-Logitech_USB_Laser_Mouse-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 May 27 13:43 usb-Razer_Razer_Huntsman_Elite-event-if01 -> ../event13
lrwxrwxrwx 1 root root   9 May 27 13:43 usb-Razer_Razer_Huntsman_Elite-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 May 27 13:43 usb-Razer_Razer_Huntsman_Elite-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 May 27 13:43 usb-Razer_Razer_Huntsman_Elite-if02-event-mouse -> ../event14
lrwxrwxrwx 1 root root   9 May 27 13:43 usb-Razer_Razer_Huntsman_Elite-if02-mouse -> ../mouse1

z3ntu, 2020-06-12:
Can you please give me the output of `sudo dkms install openrazer-driver/2.8.0` ?

wahein, 2022-03-20:
I am having this exact error message come up on Ubuntu for my Razer  Blade Stealth. Any solution found by any chance?

z3ntu, 2022-03-26:
Razer Huntsman Elite (1532:0226) is already supported, closing.

@wahein If you experience problems please open a new issue and give some details there.

closed 2022-03-26 by z3ntu