openrazer/openrazer #1159: Support for Razer Blade 15 Advanced Model (Early 2020, with OLED display)

nerdo, 2020-05-31:
### Device Info

Name was not found under the device. Since it did not appear there, I grabbed it from the sticker on the box instead.

* Name: Razer Blade 15 Advanced Model [Early 2020 model]
* Product Number: RZ09-03305E53
* Model Number: RZ09-0330

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```
Bus 001 Device 003: ID 1532:0253 Razer USA, Ltd Razer Blade
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0253
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

```
total 0
drwxr-xr-x 2 root root 200 May 31 10:44 .
drwxr-xr-x 4 root root 880 May 31 11:04 ..
lrwxrwxrwx 1 root root  10 May 31 10:44 usb-Azurewave_Integrated_Camera_200901010001-event-if00 -> ../event11
lrwxrwxrwx 1 root root  10 May 31 10:44 usb-Azurewave_Integrated_Camera_200901010001-event-if02 -> ../event12
lrwxrwxrwx 1 root root  10 May 31 10:44 usb-ELAN_Touchscreen-event-if00 -> ../event30
lrwxrwxrwx 1 root root  10 May 31 10:44 usb-Razer_Razer_Blade-event-if01 -> ../event26
lrwxrwxrwx 1 root root  10 May 31 10:44 usb-Razer_Razer_Blade-event-kbd -> ../event24
lrwxrwxrwx 1 root root  10 May 31 10:44 usb-Razer_Razer_Blade-if01-event-kbd -> ../event25
lrwxrwxrwx 1 root root  10 May 31 10:44 usb-Razer_Razer_Blade-if02-event-mouse -> ../event29
lrwxrwxrwx 1 root root   9 May 31 10:44 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

Will provide upon request if necessary.

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
[image]

j-maynard, 2020-06-01:
Can I upvote this please and if you need anything not provided let me know

marcpaskell, 2020-06-11:
+1 (base model)

j-maynard, 2020-07-16:
@nerdo Did you see a PCAP has been requested?  Are you still up for doing this?  Either that or I shall have to learn to do it myself

TheButlah, 2020-07-17:
Would the advanced model but with the 300hz screen be the same, or should I open a new issue?

nerdo, 2020-07-17:
> 
> 
> @nerdo Did you see a PCAP has been requested? Are you still up for doing this? Either that or I shall have to learn to do it myself

I did not see it was requested. Thanks for the mention. I'll try and carve out some time to get it done.

j-maynard, 2020-07-18:
> Would the advanced model but with the 300hz screen be the same, or should I open a new issue?

Have you compared the lsusb results from your laptop with the ones above to see if they are the same?

TheButlah, 2020-07-19:
I've added packet captures to #1201 and a working (I think?) pull request at #1202 . Please give the pull request a try so that we can see if things work or not