openrazer/openrazer #1161: Naga Chroma mouse intermittently unresponsive

kevchu3, 2020-06-03:
### Distribution

Fedora 32
Kernel 5.6.14-300.fc32.x86_64

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Distribution

Fedora 32
Kernel 5.6.14-300.fc32.x86_64

I'm running openrazer and my Naga Chroma mouse is unresponsive every few minutes for a few seconds at a time.  It happens after I start mousing and am in the middle of movement.  It's pretty annoying as the most important part of a mouse is being able to mouse.  My mouse changes colors, and I've noticed that the colors continue to change even as the mouse pointer freezes during that time, so I don't think it's a 'mouse went dead' thing.  I have a Lenovo Thinkpad with a red button "mouse" and it continues to be responsive (and is the only way I can mouse while my mouse is frozen).

Based on the troubleshooting wiki, I've searched through the usual logs and couldn't find anything.  For reference, I observed a number of mouse freezes after 9:47 am.

Latest logs from ~/.local/share/openrazer/logs/razer.log:
```
2020-06-03 09:47:27 | razer                          | WARNING  | Removing 0003:1532:0053.0001
2020-06-03 09:47:33 | razer                          | INFO     | Found valid device.0: 0003:1532:0053.0004
2020-06-03 09:47:33 | razer.device0                  | INFO     | Initialising device.0 RazerNagaChroma
```

The `journalctl` and `dmesg` logs don't have anything corresponding to the times of the mouse freeze either.  Is there a debug mode that I can set openrazer to, in order to gather more information?

z3ntu, 2020-06-12:
Does this issue also happen without openrazer installed?

kevchu3, 2020-06-16:
Thanks @chabad360 I've turned on my debug logs

@z3ntu I haven't been able to reproduce it recently, although it was definitely frequent when I originally opened this issue.  About a week ago, I reinstalled openrazer when I pulled in some commits from development branches, and the mouse has been a lot more stable and not freezing since.  For now, I'm going to close this out.  If the mouse unresponsiveness does reoccur frequently enough to troubleshoot, I'll post relevant debug logs.

closed 2020-06-16 by kevchu3

saberman888, 2020-12-21:
I have this issue, but it goes away once you set polling rate to 1000. While that work around seems to fix it; it could cause an issue if games like Rainbow Six Siege become playable because, it tends not to work well on 1000hz.